intel/media-driver #1777: [Bug]: VA-API cannot be used for H264 video on Dell XPS 13 9360

paulmenzel, 2024-02-21:
### Is it regression? Good in old configuration?

None

### What happened?

Debian sid/unstable with *intel-media-va-driver-non-free* 24.1.0+ds1-1 and *libigdgmm12* 22.3.17+ds1-1

Use [Bigbuck Bunny 8Bit](https://avtshare01.rz.tu-ilmenau.de/avt-vqdb-uhd-1/test_1/segments/bigbuck_bunny_8bit_15000kbps_1080p_60.0fps_h264.mp4) from [AVT-VQDB-UHD-1 video quality database](https://telecommunication-telemedia-assessment.github.io/AVT-VQDB-UHD-1/test_1.html):

> Test 1 File
> filename extension: mp4
> bit rate: 14794.94
> target bit rate: 15000
> height: 1080
> width: 1920
> frame rate: 60.0
> codec: h264
> MOS: 4.20689655172414
> CI: 0.281571835611333

mpv does not use VA-API and drops frames:

    $ mpv --hwdec=vaapi bigbuck_bunny_8bit_15000kbps_1080p_60.0fps_h264.mp4 
     (+) Video --vid=1 (*) (h264 1920x1080 60.000fps)
    VO: [gpu] 1920x1080 yuv422p
    (Paused) V: 00:00:05 / 00:00:10 (52%) Dropped: 136

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

```
$ sudo dmesg | grep -e "DMI:" -e "Linux version" -e microcode
[    0.000000] Linux version 6.8.0-rc4+ (build@bohemianrhapsody.molgen.mpg.de) (gcc (Debian 13.2.0-13) 13.2.0, GNU ld (GNU Binutils for Debian) 2.42) #25 SMP PREEMPT_DYNAMIC Sat Feb 17 05:39:03 CET 2024
[    0.000000] DMI: Dell Inc. XPS 13 9360/0596KF, BIOS 2.21.0 06/02/2022
[    0.367292] microcode: Current revision: 0x000000f4
[    0.367293] microcode: Updated early from: 0x000000f0
$ lspci -nn -s 02.0
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 620 [8086:5916] (rev 02)
02:02.0 PCI bridge [0604]: Intel Corporation DSL6340 Thunderbolt 3 Bridge [Alpine Ridge 2C 2015] [8086:1576]
$ vainfo
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
$ mpv --version
mpv 0.37.0 Copyright © 2000-2023 mpv/MPlayer/mplayer2 projects
libplacebo version: v6.338.2
FFmpeg version: 6.1.1-1
FFmpeg library versions:
   libavutil       58.29.100
   libavcodec      60.31.102
   libavformat     60.16.100
   libswscale      7.5.100
   libavfilter     9.12.100
   libswresample   4.12.100

```

### Do you want to contribute a patch to fix the issue?

No.

paulmenzel, 2024-02-21:
VA-API normally can be used by mpv, but these test files seem to pose problems. VA-API is *not* used with lower resolutions either:

```
$ wget https://avtshare01.rz.tu-ilmenau.de/avt-vqdb-uhd-1/test_1/segments/bigbuck_bunny_8bit_750kbps_360p_60.0fps_h264.mp4
--2024-02-21 10:18:48--  https://avtshare01.rz.tu-ilmenau.de/avt-vqdb-uhd-1/test_1/segments/bigbuck_bunny_8bit_750kbps_360p_60.0fps_h264.mp4
Auflösen des Hostnamens avtshare01.rz.tu-ilmenau.de (avtshare01.rz.tu-ilmenau.de)… 141.24.217.246
Verbindungsaufbau zu avtshare01.rz.tu-ilmenau.de (avtshare01.rz.tu-ilmenau.de)|141.24.217.246|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 972959 (950K) [video/mp4]
Wird in »bigbuck_bunny_8bit_750kbps_360p_60.0fps_h264.mp4« gespeichert.

bigbuck_bunny_8bit_750kbps_360p_60.0fps_h26 100%[===========================================================================================>] 950,16K  5,90MB/s    in 0,2s    

2024-02-21 10:18:48 (5,90 MB/s) - »bigbuck_bunny_8bit_750kbps_360p_60.0fps_h264.mp4« gespeichert [972959/972959]

$ mpv bigbuck_bunny_8bit_750kbps_360p_60.0fps_h264.mp4 
 (+) Video --vid=1 (*) (h264 640x360 60.000fps)
VO: [gpu] 640x360 yuv422p
Exiting... (End of file)
```

Jexu, 2024-03-04:
Do you check if ffmpeg or sample_decode could decode such clip? I don't know whether mpv has this strategy that forwards it to sw decoder for small resolution clip?

paulmenzel, 2024-06-16:
> Do you check if ffmpeg or sample_decode could decode such clip?

No, it can’t:

```
$ ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i bigbuck_bunny_8bit_15000kbps_1080p_60.0fps_h264.mp4 -f null -
ffmpeg version 6.1.1-4+b3 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13 (Debian 13.2.0-25)
  configuration: --prefix=/usr --extra-version=4+b3 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --disable-omx --enable-gnutls --enable-libaom --enable-libass --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libharfbuzz --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-openal --enable-opencl --enable-opengl --disable-sndio --enable-libvpl --disable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-ladspa --enable-libbluray --enable-libjack --enable-libpulse --enable-librabbitmq --enable-librist --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libx264 --enable-libzmq --enable-libzvbi --enable-lv2 --enable-sdl2 --enable-libplacebo --enable-librav1e --enable-pocketsphinx --enable-librsvg --enable-libjxl --enable-shared
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'bigbuck_bunny_8bit_15000kbps_1080p_60.0fps_h264.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf57.56.100
  Duration: 00:00:10.00, start: 0.033008, bitrate: 15156 kb/s
  Stream #0:0[0x1](und): Video: h264 (High 4:2:2) (avc1 / 0x31637661), yuv422p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 15150 kb/s, 60 fps, 60 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf60.16.100
  Stream #0:0(und): Video: wrapped_avframe, yuv422p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 60 fps, 60 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 wrapped_avframe
[out#0/null @ 0x558971f46bc0] video:281kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
frame=  600 fps=263 q=-0.0 Lsize=N/A time=00:00:09.98 bitrate=N/A speed=4.38x
```

Current failing test with mpv:

```
$ mpv --version
mpv 0.38.0 Copyright © 2000-2024 mpv/MPlayer/mplayer2 projects
libplacebo version: v6.338.2
FFmpeg version: 6.1.1-4+b3
FFmpeg library versions:
   libavutil       58.29.100
   libavcodec      60.31.102
   libavformat     60.16.100
   libswscale      7.5.100
   libavfilter     9.12.100
   libswresample   4.12.100

$ mpv -v -v -v --vo=gpu --hwdec=vaapi bigbuck_bunny_8bit_15000kbps_1080p_60.0fps_h264.mp4
[…]
[vo/gpu] Assuming 59.981809 FPS for display sync.
[vd] Container reported FPS: 60.000000
[vd] Codec list:
[vd]     h264 - H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
[vd]     h264_v4l2m2m (h264) - V4L2 mem2mem H.264 decoder wrapper
[vd]     h264_qsv (h264) - H264 video (Intel Quick Sync Video acceleration)
[vd]     h264_cuvid (h264) - Nvidia CUVID H264 decoder
[... 40 lines cut ...]
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

thesword53, 2024-11-04:
> `VO: [gpu] 1920x1080 yuv422p`

The video you are trying to decode is encoded with H.264 codec and **yuv422p** pixel format, but the corresponding profile isn't supported by VAAPI. Usually most of videos are encoded using **yuv420p** pixel format.

paulmenzel, 2024-11-04:
Thank you for pointing this out.

paulmenzel, 2024-11-04:
> > `VO: [gpu] 1920x1080 yuv422p`
> 
> The video you are trying to decode is encoded with H.264 codec and **yuv422p** pixel format, but the corresponding profile isn't supported by VAAPI. Usually most of videos are encoded using **yuv420p** pixel format.

Is there a list of profiles supported by VA-API?

closed 2024-11-04 by paulmenzel