intel/media-driver #1779: [Feature]: Recognize allow_content_tools and support force_integer_mv

rosetta-jpn, 2024-03-12:
### What Feature?

Libva patch to configure allow_content_tools and force_integer_mv is merged.
https://github.com/intel/libva/pull/795

If [allow_screen_content_tools](https://aomediacodec.github.io/av1-spec/#uncompressed-header-semantics) is set to true, the decoder reads [palette mode info](https://aomediacodec.github.io/av1-spec/#palette-mode-info-syntax) in [intra_frame_mode_info()](https://aomediacodec.github.io/av1-spec/#intra-frame-mode-info-syntax) and [intra_block_mode_info()](https://aomediacodec.github.io/av1-spec/#intra-block-mode-info-syntax).
If palette mode is not enabled, then the encoder needs to fill has_palette_y and has_platte_uv to false.
So the driver need to recognize allow_content_tools (and palette mode) to produce a valid stream.

In addition, I would also like to ask adding force_integer_mv support to media-driver. 

Thanks in advance.

### What's the usage scenario would be benifited?

Video Conference, Content Creation

### Do you want to contribute a patch to develop this feature?

None

rosetta-jpn, 2024-03-25:
@XinfengZhang Do you have any timeline for this?

XinfengZhang, 2024-03-26:
hi @rosetta-jpn , you could do some quick evaluation with hard code to enable integer MV
https://github.com/intel/media-driver/blob/master/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/codec/hal/enc/av1/features/encode_av1_vdenc_const_settings_xe_lpm_plus_base.cpp#L387
change the value to 0xc0a080 to have a try. 

to be safe, maybe also need add params.forceIntegerMv = 1; in  https://github.com/intel/media-driver/blob/master/media_softlet/agnostic/common/codec/hal/enc/av1/features/encode_av1_basic_feature.cpp#L880

rosetta-jpn, 2024-03-28:
@XinfengZhang I tried on MeteoLake.

> change the value to 0xc0a080 to have a try.

This doesn't make a difference on produced bitstream.

> maybe also need add params.forceIntegerMv = 1

This makes the produced bitstream invalid (failed in decoding).

XinfengZhang, 2024-03-29:
hi hiro, could you help dump the  encoding command buffer and share it.
BKM https://github.com/intel/media-driver/wiki/Decode-and-Encode-Debug-Tool-and-surface-dump

rosetta-jpn, 2024-04-08:
Nothing is output. I guess I don't build my driver in neither debug nor release-internal. 
Could you tell me how to build the debug version driver?

zlma7001, 2024-04-22:
Hiro, we have verified PR https://github.com/intel/media-driver/pull/1794, and it hard code to force use integer MV if you would like to have a try, and I will paste the data for av1 vaapi encoder later.