intel/media-driver #1783: [Bug]: ENABLE_XE_KMD not active by default.

fmc000, 2024-03-23:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

None

### What happened?

Activating the "xe" kernel module and using the latest Mesa that have this driver enabled breaks hardware video acceleration. 3D and vulkan apps work just fine.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

vainfo:
```
Trying display: wayland
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 18 (invalid parameter),exit
```

cvlc:
```
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[00005bd3e58f5340] dummy interface: using the dummy interface module...
libva error: /usr/lib/dri/iHD_drv_video.so init failed
[00007f3878001ec0] glconv_vaapi_x11 gl error: vaInitialize: invalid parameter
libva error: /usr/lib/dri/iHD_drv_video.so init failed
[00007f3878001ec0] glconv_vaapi_drm gl error: vaInitialize: invalid parameter
Xlib:  extension "DRI2" missing on display ":1".
libva error: vaGetDriverNames() failed with operation failed
[00007f3878001ec0] glconv_vaapi_drm gl error: vaInitialize: operation failed
```

### Debug Information

Please check https://bbs.archlinux.org/viewtopic.php?pid=2158940  

Building with -DENABLE_XE_KMD=ON fixes the problem for me

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-03-25:
Please update you code version. It has been enabled by default from commit dbcf5ef421b24c9cebdcce41a96093e63a0088aa

XinfengZhang, 2024-03-25:
@Jexu , suppose you need add such code in cmake
option (ENABLE_XE_KMD "Enable Production KMD header files" ON)

PR #1784 (2024-03-26): Enable Xe KMD by default
fix #1783
closed 2024-03-26 by intel-mediadev