intel/media-driver #1788: [Bug]: Gstreamer link is dead

SuperSandro2000, 2024-04-04:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

The link at https://github.com/intel/media-driver/wiki/Gstreamer is no longer working and redirects to a generic overview site.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-04-09:
Hi @Sherry-Lin could you help update wiki.

Sherry-Lin, 2024-04-09:
@xhaihao  do you know the new location for ffmpeg/gst setup? I will update them in the wiki page  https://github.com/intel/media-driver/wiki/Gstreamer and https://github.com/intel/media-driver/wiki/FFMpeg. Thanks.