intel/media-driver #1789: [Bug]: OBS 30.1.1 crashes with double free error if VAAPI is enabled and there are two Intel GPUs present

PaddyMac, 2024-04-06:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

I have a Dell Optiplex 7010 with an i7-3770S processor which has integrated Intel graphics. I had an AMD R7 240 graphics card but recently upgraded to an Intel Arc A380 graphics card. I moved the R7 to a secondary PCIE slot and put the A380 in the primary PCIE slot. I have a dual monitor setup. My primary monitor is plugged into the A380 via a HDMI to DVI-D cable, and my secondary monitor is (or was) plugged into the onboard VGA port. And this appears to be the crux of the issue.

I recently upgraded to OBS Studio 30.1.1 which has better support for Intel QSV. After I upgraded my AMD R7 240 to the A380, OBS began crashing at start with a double free error. After three days of frustration and much trial and error, the conclusion I came to is that the problem is with the Intel Media driver. OBS stopped crashing when I unplugged my secondary monitor and rebooted because this disables the onboard graphics. I plugged my secondary monitor into the AMD R7 240 and have no problems.

Apparently the Intel Media driver does not handle multiple Intel graphics devices being present at the same time -- perhaps especially if they are different generations of devices. I'm not sure how relevant this is, but OBS crashed when it was checking VAAPI support, so it may be (probably is) VAAPI-related.

I run Gentoo Linux, and I ran OBS through gdb, but I didn't see anything obvious that wasn't already shown in the console output. That said, I'm not a programmer. But I can follow instructions if you need additional information. My instinct tells me that the double free may be a result of either Intel Media driver performing an action twice because two devices are present or that there are two different Intel drivers performing the same action and conflicting with each other.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

The intent will be game streaming via OBS, but since OBS is crashing at start due to this bug rather than during OBS use, that isn't relevant except that it is preventing me from starting OBS to stream.

### Debug Information

libva 2.21.0
libva utils 2.20.1
gmmlib 22.3.18
media driver 2.4.1

/dev/dri does show all GPUs present (2 as of this moment as the onboard graphics are currently disabled which, if enabled, would show 3)

03:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A380] [8086:56a5] (rev 05)
06:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Oland PRO [Radeon R7 240/340 / Radeon 520] [1002:6613] (rev 87)

libva info: VA-API version 1.21.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/va/drivers/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
Trying display: wayland
vainfo: VA-API version: 1.21 (libva 2.20.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.2.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP

The last relevant debug info requested will require rebooting and rearranging some cables, so I'll have to add that later.

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-04-09:
Please paste the crash log message if you have it. And the crash happens in media driver, libva or vpl rt?

eero-t, 2024-04-24:
Does either `vainfo` or `vpl-inspect` command crash?

If not, and you do not have backtrace [1] pointing crash to be originating from one of the Intel media libraries, this looks more like OBS not supporting multiple VAAPI drivers listing GPUs (i.e. you should file it against OBS).

[1] Some alternatives for getting backtraces:
* easy, but lots of output:  `strace -f -k <app>`
* 30x slowdown, but much better backtraces: `valgrind <app>`
* allows further debugging: start `gdb`, and `run` application from it

(On most distros, above tools come from packages with the same name.)

PaddyMac, 2024-05-02:
I apologize for not responding sooner. Only today did I have an opportunity to give this the time and attention needed. I ran all the requested tests, and I am attaching the output for the various tests. It *did* crash when I ran vpl-inspect. Oddly enough, it did *not* crash when I ran OBS via Valgrind. Therefore I am not attaching anything from valgrind. I also am unable to attach the output from strace because the log file is about 46 MiB in size.

[gdb.txt](https://github.com/intel/media-driver/files/15194660/gdb.txt)
[vainfo.txt](https://github.com/intel/media-driver/files/15194668/vainfo.txt)
[valgrind.txt](https://github.com/intel/media-driver/files/15194670/valgrind.txt)
[vpl-inspect.txt](https://github.com/intel/media-driver/files/15194671/vpl-inspect.txt)

PaddyMac, 2024-05-02:
I unintentionally attached a file for valgrind above, but it was just the console output from OBS. Here is the actual Valgrind output after properly directing it to a log file.

[valgrind.txt](https://github.com/intel/media-driver/files/15194880/valgrind.txt)

eero-t, 2024-05-03:
Valgrind output shows both the intial alloc and the extra free to happen within media-driver (not in VA-API library):
```
==30669== Invalid read of size 8
==30669==    at 0x35ABBFC5: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35ABBD01: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x359064C9: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x3601E56A: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35A7EE25: __vaDriverInit_1_21 (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0xA6EC7E1: vaInitialize (in /usr/lib64/libva.so.2.2100.0)
...
==30669==  Address 0x14f88930 is 1,600 bytes inside a block of size 63,544 free'd
==30669==    at 0x484395F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==30669==    by 0x35D4A275: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35A3E27D: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35ABBC08: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x359064C9: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x3601E56A: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35A7EE25: __vaDriverInit_1_21 (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0xA6EC7E1: vaInitialize (in /usr/lib64/libva.so.2.2100.0)
...
==30669==  Block was alloc'd at
==30669==    at 0x4847E43: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==30669==    by 0x35ABBB28: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x359064C9: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x3601E56A: ??? (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0x35A7EE25: __vaDriverInit_1_21 (in /usr/lib64/va/drivers/iHD_drv_video.so)
==30669==    by 0xA6EC7E1: vaInitialize (in /usr/lib64/libva.so.2.2100.0)
==30669==    by 0x37C523CE: check_adapter(void*, char const*, unsigned int) (in /usr/lib64/obs-plugins/obs-qsv11.so)
```

What I assume to have happened in this case is that libva calls media driver init twice, on one of the calls media driver recognizes the GPU, on another, it does not, and it does not handle that correctly.

This is a bit odd as media driver is tested and works on setups with multiple Intel dGPUs, so multiple init calls succeeding should not be a problem.

eero-t, 2024-05-03:
I was able to reproduce the issue with following stack:
- libva:    2.20.0
- GMMlib:   intel-gmmlib-22.3.17
- Media:    intel-media-24.1.2

But not any more with a newer one than yours:
- libva:    2.21.0
- GMMlib:   intel-gmmlib-22.3.19
- Media:    intel-media-24.2.1

=> @PaddyMac Could you try upgrading?

This was on server with a non-Intel iGPU and several Intel dGPUs:
```
# head -3 /sys/class/drm/card?/device/uevent
==> /sys/class/drm/card0/device/uevent <==
DRIVER=ast
PCI_CLASS=30000
PCI_ID=1A03:2000

==> /sys/class/drm/card1/device/uevent <==
DRIVER=i915
PCI_CLASS=38000
PCI_ID=8086:56C0

==> /sys/class/drm/card2/device/uevent <==
DRIVER=i915
PCI_CLASS=38000
PCI_ID=8086:56C0
```

Note: `ast`  iGPU offers only `card0` DRM node, not `renderXXX` one.

When testing with HadesCanyon (KabyLake) NUC that had in same package both Intel KBL iGPU and AMD Vega dGPU:
```
# head -3 /sys/class/drm/card?/device/uevent
==> /sys/class/drm/card0/device/uevent <==
DRIVER=i915
PCI_CLASS=38000
PCI_ID=8086:591B

==> /sys/class/drm/card1/device/uevent <==
DRIVER=amdgpu
PCI_CLASS=30000
PCI_ID=1002:694C
```

Issue was not reproducible. I guess the reason was Intel GPU being before non-Intel one.

=> @PaddyMac If you cannot upgrade the driver, or upgrade does not help, could you try whether double free goes away if you have Intel GPU before AMD one?

eero-t, 2024-05-03:
> => @PaddyMac If you cannot upgrade the driver, or upgrade does not help, could you try whether double free goes away if you have Intel GPU before AMD one?

Ah, read the bug description closer.  Bug is triggered by i7-3770S being IvyBridge, i.e. having Intel iGPU that is not supported by this Intel driver project (only by legacy i965 media driver), i.e. same case as non-Intel GPU.  And as it's iGPU, if it's enabled, it will be the first found GPU, i.e. one will need a fixed media-driver.

Jexu, 2024-05-09:
It looks like same issue in https://github.com/intel/media-driver/issues/1795

When trying to open media driver unssuccefully, then media driver will exit abnormally since failing to handle error. This should be fixed in media driver.

In your case, monitor connected to an iGPU that media driver doesn't support, so we expect it exits media driver normally without any crash.

eero-t, 2024-05-10:
> It looks like same issue in #1795

Tthat one happens also with media driver v24.2.1, whereas in my testing this issue seems to be fixed in that release, see https://github.com/intel/media-driver/issues/1789#issuecomment-2092705816.

@PaddyMac Can you confirm whether your issue is fixed by upgrading to v2.4.1 (or newer) media driver: https://github.com/intel/media-driver/tags ?

MicroYY, 2024-05-28:
Most probably the culprit code is https://github.com/intel/media-driver/blob/a0391f7a051d566987d6db0fb026db5ab45f3bae/media_softlet/linux/common/os/i915/mos_bufmgr.c#L5648
```
int mos_get_device_id(int fd, uint32_t *deviceId)
{
    int device_type = mos_query_device_type(fd);

    if (DEVICE_TYPE_I915 == device_type)
    {
        return mos_get_dev_id_i915(fd, deviceId);
    }
#ifdef ENABLE_XE_KMD
    else if (DEVICE_TYPE_XE == device_type)
    {
        return mos_get_dev_id_xe(fd, deviceId);
    }
#endif
    return -ENODEV;
}
```
here return a negative value which is used in https://github.com/intel/media-driver/blob/a0391f7a051d566987d6db0fb026db5ab45f3bae/media_softlet/linux/common/os/hwinfo_linux.cpp#L70
```
if (mos_get_device_id(fd, &devId))
{
    MOS_OS_ASSERTMESSAGE("Failed to get the chipset id\n");
    return MOS_STATUS_INVALID_HANDLE;
}
```
so this branch will never be bit...

I don't have systems with both intel GPU and non-intel GPU. 
You may try this pull request https://github.com/intel/media-driver/pull/1805
If it won't help, very appreciated if gdb crash log from release-internal build is provided. (build with 
**-DBUILD_TYPE=release-internal**)

Jexu, 2024-05-28:
@MicroYY
Driver returns error code here doesn't matter since it opens an unsupported igpu. Problem is that driver could not exit with all resource released and sw stack crash happens in media driver. In this case, application may have no chance to handle error.

intel-mediadev, 2024-05-28:
Auto Created VSMGWL-73921 for further analysis.

eero-t, 2024-05-28:
> I don't have systems with both intel GPU and non-intel GPU.

@MicroYY This bug is reported for a setup with **only** Intel GPUs.  Both of them supported by Intel KMD, but first one not being supported by the media driver.

You can create similar setup by using older GEN host with iGPU and building media driver version where support for that GEN is disabled.    Or by having 2 different dGPUs, assigning first one to Xe KMD and second to i915, and building media driver without Xe KMD support.

MicroYY, 2024-05-30:
I guess I reproduced it on a adl+dg2 system via `vpl-inspect`...
`vpl-inspect` opens iGPU first and then dGPU. When querying engines, to be exact, vebox engine, of dGPU, no vebox is got from `Ioctl`. Media driver will crash with double free because of vebox context creation failure.

[image]
Modified code of `mos_bufmgr_get_driver_info`:
```
    drvInfo->hasVebox = 0;
    retValue = 0;
    if (mos_get_param(fd, I915_PARAM_HAS_VEBOX, &retValue))
    {
        drvInfo->hasVebox = !!retValue;
        printf("fd %d has vebox %u\n", fd, retValue);
    }
```

MicroYY, 2024-05-30:
I investigated more and here is a summary. The double free issue is reproduceable on i+d system where dGPU vebox engine cannot be queried from KMD. In such case vebox context on dGPU cannot be created and some allocations were double-freed by media driver. 
A right expectation is vebox being queried from KMD and context being created successfully on dGPU. KMD could have some issues on exposing hw engine. Nevertheless, media driver should handle whatever cases. 
Updated https://github.com/intel/media-driver/pull/1805 to fix double free crash.

closed 2024-06-21 by MicroYY