intel/media-driver #1793: [Bug]: Transcoding using jellyfin-ffmpeg5 causes a system crash

kropop, 2024-04-21:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

Transcoding in Jellyfin using hardware acceleration causes a system crash that requires a hard reboot. Before the crash, the performace figures are great - transcoding a H264 stream runs at over 60 FPS, manages to transconde a minute or two of the video.
The CPU is Intel Celeron J3455

It happens both in docker and on bare metal configuration.

Software (CPU) decoding works fine, although for obvious reasons, has poor performance.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

1. docker exec 2407f7e9e127 /usr/lib/jellyfin-ffmpeg/vainfo
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
Trying display: drm
vainfo: VA-API version: 1.20 (libva 2.20.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.4.0 (05e4d0d)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD

2. docker exec 2407f7e9e127 ls /dev/dri
card0
renderD128

dmesg doesn't seem to have any hardware specific information about any crashes, neither does journalctl.

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2024-04-22:
How about the ffmpeg transcoding? Could it reproduce by ffmpeg-vaapi or ffmpeg-qsv?

kropop, 2024-04-24:
> > Transcoding in Jellyfin using hardware acceleration causes a system crash that requires a hard reboot.
> 
> What _exactly_ is meant by system crash?
> 
> If you mean that UI is not any more usable, that could also be a bug somewhere in the graphics stack.
> 
> If your machine stops responding e.g. to SSH, that's likely a kernel crash i.e. nothing to do with any user-space driver, like this one.
> 
The machine stops responding to SSH and the connection terminates after a short while. All running CLI applications also stop responding - btop, intel_gpu_top (I used them to monitor if the hardware acceleration is working properly). If it's not the driver's fault, then that doesn't sound too good for me..
> Which kernel version you're using and does it have any drivers in it that are not from your distribution?
> 
It's running Debian Bookworm stable using kernel version 6.1.0-20-amd64.
> Does crash happen also with the latest kernel?
> 
It's already on the latest kernel.
> > dmesg doesn't seem to have any hardware specific information about any crashes, neither does journalctl.
> 
> dmesg is only from current boot, so it will not have anything from before reboot. If this is hard kernel crash, one may need network (or serial) console to catch kernel error messages.
How can I record those error messages if that's the case?

eero-t, 2024-04-24:
> The machine stops responding to SSH and the connection terminates after a short while. 

Ok, that's a kernel, not a user-space media driver bug.  You need to report this against kernel driver.

Assuming it is indeed Intel driver bug, the correct project is this one: https://gitlab.freedesktop.org/drm/i915/kernel/-/issues

> It's running Debian Bookworm stable using kernel version 6.1.0-20-amd64.
>
>> Does crash happen also with the latest kernel?
> It's already on the latest kernel.

From kernel developers point of view, 6.1 is rather old, latest upstream kernel is 6.8, with 6.9 being released soon.

kropop, 2024-04-26:
I reinstalled my entire setup with Ubuntu LTS and the problem is gone.

closed 2024-04-26 by kropop