intel/media-driver #1796: [Bug]: Gstreamer vaapipostproc element converts ARGB format incorrectly

russnic, 2024-04-30:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

When I attempt to create a test video via gstreamer using ARGB format the `vaapipostproc` does not convert the colours correctly, see below.

ARGB Pipeline:

```
gst-launch-1.0 -e videotestsrc num-buffers=60 ! video/x-raw,format=ARGB,width=1920,height=1080,framerate=60/1 ! vaapipostproc ! vaapih264enc ! h264parse ! mp4mux ! filesink location=/home/user/videotestsrc-argb.mp4
```

ARGB Video:

https://github.com/intel/media-driver/assets/128496652/2c4e659c-494b-4fda-813e-98cd0eaefe68

A pipeline that generates a test video with UYVY format and vaapipostproc has the correct colours

UYVY Pipeline:

```
gst-launch-1.0 -e videotestsrc num-buffers=60 ! video/x-raw,format=UYVY,width=1920,height=1080,framerate=60/1 ! vaapipostproc ! vaapih264enc ! h264parse ! mp4mux ! filesink location=/home/user/videotestsrc-uyvy.mp4
```

UYVY Video:

https://github.com/intel/media-driver/assets/128496652/086d0fef-5c4a-46e7-a259-ed7713161bbe

A pipeline that swaps `vaapipostproc` for `videoconvert` using ARGB format has the correct colours

`videoconvert` pipeline:

```
gst-launch-1.0 -e videotestsrc num-buffers=60 ! video/x-raw,format=ARGB,width=1920,height=1080,framerate=60/1 ! videoconvert ! vaapih264enc ! h264parse ! mp4mux ! filesink location=/home/user/videotestsrc-videoconvert-argb.mp4
```

https://github.com/intel/media-driver/assets/128496652/cf8355e8-7405-42b9-b943-492bd9074bf9

### What's the usage scenario when you are seeing the problem?

Content Creation

### What impacted?

All of the above was completed on Ubuntu 24.04 and GStreamer 1.24.2 

However, I've also tested/confirmed issue is on Ubuntu 20.04 and GStreamer 1.16.3

### Debug Information

1. 
I’m unable to find `libva/libva-utils/gmmlib/media-driver` on my system but have the below installed
```
user@host:~$ apt list --installed intel*
Listing... Done
intel-media-va-driver-non-free/noble,now 24.1.0+ds1-1 amd64 [installed]
```

2. 
```
user@host:~$ ls /dev/dri
by-path  card0  renderD128
```

3. 
```
user@host:~$ lspci -nn |grep  -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620] [8086:3ea0] (rev 02)
```

4. 
```
user@host:~$ sudo vainfo
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :    VAEntrypointVideoProc
      VAProfileNone                   :    VAEntrypointStats
      VAProfileMPEG2Simple            :    VAEntrypointVLD
      VAProfileMPEG2Simple            :    VAEntrypointEncSlice
      VAProfileMPEG2Main              :    VAEntrypointVLD
      VAProfileMPEG2Main              :    VAEntrypointEncSlice
      VAProfileH264Main               :    VAEntrypointVLD
      VAProfileH264Main               :    VAEntrypointEncSlice
      VAProfileH264Main               :    VAEntrypointFEI
      VAProfileH264Main               :    VAEntrypointEncSliceLP
      VAProfileH264High               :    VAEntrypointVLD
      VAProfileH264High               :    VAEntrypointEncSlice
      VAProfileH264High               :    VAEntrypointFEI
      VAProfileH264High               :    VAEntrypointEncSliceLP
      VAProfileVC1Simple              :    VAEntrypointVLD
      VAProfileVC1Main                :    VAEntrypointVLD
      VAProfileVC1Advanced            :    VAEntrypointVLD
      VAProfileJPEGBaseline           :    VAEntrypointVLD
      VAProfileJPEGBaseline           :    VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:    VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:    VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:    VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:    VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :    VAEntrypointVLD
      VAProfileVP8Version0_3          :    VAEntrypointEncSlice
      VAProfileHEVCMain               :    VAEntrypointVLD
      VAProfileHEVCMain               :    VAEntrypointEncSlice
      VAProfileHEVCMain               :    VAEntrypointFEI
      VAProfileHEVCMain10             :    VAEntrypointVLD
      VAProfileHEVCMain10             :    VAEntrypointEncSlice
      VAProfileVP9Profile0            :    VAEntrypointVLD
      VAProfileVP9Profile2            :    VAEntrypointVLD
```

5. 
ARGB:
[libva_trace.log.142533.thd-0x00179cb4.txt](https://github.com/intel/media-driver/files/15160632/libva_trace.log.142533.thd-0x00179cb4.txt)
[libva_trace.log.142533.thd-0x00179ccf.txt](https://github.com/intel/media-driver/files/15160633/libva_trace.log.142533.thd-0x00179ccf.txt)
[libva_trace.log.142533.thd-0x00179cd1.txt](https://github.com/intel/media-driver/files/15160634/libva_trace.log.142533.thd-0x00179cd1.txt)

UYVY:
[libva_trace.log.142701.thd-0x00179d8a.txt](https://github.com/intel/media-driver/files/15160646/libva_trace.log.142701.thd-0x00179d8a.txt)
[libva_trace.log.142701.thd-0x00179d8d.txt](https://github.com/intel/media-driver/files/15160647/libva_trace.log.142701.thd-0x00179d8d.txt)
[libva_trace.log.142701.thd-0x00179d8f.txt](https://github.com/intel/media-driver/files/15160648/libva_trace.log.142701.thd-0x00179d8f.txt)

6.
GPU does not hang

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-05-08:
Auto Created VSMGWL-73412 for further analysis.

wennibao, 2024-07-08:
hi @russnic, we have captured the input from the media-driver and observed the same color discrepancy. It appears that this is not an issue originating from the driver.

wennibao, 2024-07-08:
If there are no further concerns, I intend to close this bug with the status 'rejected'. Please do not hesitate to raise any questions you may have.

russnic, 2024-07-08:
Hi @wennibao, thank you for taking the time to look at this issue. Could you please advise where you think the issue is coming from? So I can log a bug report with them.

wennibao, 2024-07-08:
Hello @russnic,
It appears that GStreamer generates input for the media-driver using videotestsrc. I believe this issue could be reported to the GStreamer team for further investigation.

russnic, 2024-07-08:
Hi @wennibao, 

The `videotestsrc` is working as intended as the video colours output fine when using `videoconvert` instead of the `vaapipostproc`. 

Sorry I was under the assumption you own the `vaapipostproc`, or is this better suited with GStreamer?

ceyusa, 2024-07-08:
Hi @wennibao ,

This is not fixed, this is, actually, a duplicate of https://github.com/intel/media-driver/issues/644 which remains open

wennibao, 2024-07-24:
Hi @russnic, @ceyusa. We have identified a discrepancy in the format definitions between Libva and the media-driver. Our team is currently working on resolving this issue. We will keep you updated on our progress.

wennibao, 2024-07-24:
For your convenience, @russnic, you may consider using BGRA as a substitute for ARGB. Hope this will help you.

russnic, 2024-07-24:
Hi @wennibao,

Thank you for the update! 

Unfortunately in our case we cannot control the output of the source, I used the `videotestsrc` element in the ticket for ease.

wennibao, 2024-08-09:
Hi @russnic 
The driver has been updated in commit revision 7c9645864d1437480a347d7fdac502ffe21d5716. In this update, we have removed support for ARGB caps and now exclusively support the Alpha channel in the high bits. Please ensure to query the capabilities first.

wennibao, 2024-08-09:
If no more concern, I will mark this issue as resolved

russnic, 2024-08-09:
Hi @wennibao 

Looking at the commit you linked it just removes all references to ARGB, so the driver will error out now instead of converting to the wrong colours? 

Unfortunately this doesn't help me, as stated previously I cannot control the output from the source.

Do I need to log a feature request elsewhere for the driver to support ARGB, or can this issue remain open for the feature request?

wennibao, 2024-08-12:
hi @russnic, please be advised that the media driver will not support ARGB format, where the Alpha channel is in the low bits. It would be appreciated if you could inform the source producer of this change and advise them to refrain from using ARGB in future submissions.

wennibao, 2024-08-12:
Fixes: intel/media-driver#1796

MicroYY, 2024-08-13:
I'm closing this issue. Pls feel free to reopen it for any concern.

closed 2024-08-13 by MicroYY

xiaoton1, 2024-10-18:
BTW, please use vapostproc, all vaapi-xxx plugins are legacy and deprecated.

russnic, 2024-10-18:
Thanks @xiaoton1 unfortunately the same issue occurs. Do you know if you plan to support ARGB in `vapostproc`?

`
gst-launch-1.0 -e videotestsrc num-buffers=60 ! video/x-raw,format=ARGB,width=1920,height=1080,framerate=60/1 ! vapostproc ! vaapih264enc ! h264parse ! mp4mux ! filesink location=/tmp/videotestsrc-vapostproc-argb.mp4
`

https://github.com/user-attachments/assets/526888b1-0709-4dda-b9ff-67c768231b89

xiaoton1, 2024-10-19:
Hi,
This is the hw limitation. We cannot support RGB format w/ Alpha channel in high address. Wenni has submitted one PR to remove ARGB caps to avoid confusing.

Thanks!

Xiaotong