intel/media-driver #1797: [Bug]:

heitbaum, 2024-05-01:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

When compiling with gcc-14.1 (rc) media-driver 24.2.2 fails to compile with the following errors.

`error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor] `

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Unable to compile with gcc-14.1

### Debug Information

Na

### Do you want to contribute a patch to fix the issue?

No.

rubin55, 2024-05-31:
FWIW, I just tried to compile 24.2.4 using `gcc` 14.1.1, this fails with a lot of errors, here's the output log for those interested: 
[output.log](https://github.com/user-attachments/files/15510335/output.log)

MicroYY, 2024-05-31:
Fix by: https://github.com/intel/media-driver/pull/1808

commit e77144df (2024-06-06): Fix GCC14 compilation issue

Fix: #1797
closed 2024-06-06 by intel-mediadev