intel/media-driver #1799: [Bug]: 10bit RT_FORMAT are contains in VP9 profile 0 and 1 encoding

rosetta-jpn, 2024-05-10:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

See vainfo output on gen12 device. [vainfo.txt](https://github.com/intel/media-driver/files/15269759/vainfo.txt).
VAProfileVP9Profile0/VAEntrypointEncSlice contains VA_RT_FORMAT_YUV420_10BPP.
But 10 bits encoding is infeasible in vp9 profile 0 and 1.
It is good to correct them.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-05-13:
Auto Created VSMGWL-73538 for further analysis.

leyu-yao, 2024-05-21:
@rosetta-jpn  That's correct. 10 bits encoding is infeasible in vp9 profile 0 and 1, according to vp9 spec.
We would plan to fix it.