intel/media-driver #1800: [Bug]: Arc A750 (DG2) shows segfault or bus error with raw video from GStreamer ximagesrc (works fine with AMD)

ehfd, 2024-05-12:
### Which component impacted?

Encode, Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. Ubuntu 24.04, Intel Arc A750
2. Default in Ubuntu 24.04 (Kernel: 6.8.0-31-generic, intel-media-va-driver-non-free 24.1.0+ds1-1, libva 2.20.0), GStreamer 1.24.2
3. `gdb --args gst-launch-1.0 ximagesrc show-pointer=0 remote=1 blocksize=16384 use-damage=0 ! "video/x-raw,framerate=60/1" ! vapostproc scale-method=fast ! "video/x-raw(memory:VAMemory),format=NV12" ! vah264lpenc aud=false b-frames=0 cpb-size=67 dct8x8=false key-int-max=1024 mbbrc=enabled num-slices=4 rate-control=cbr target-usage=4 bitrate=4000 ! "video/x-h264,profile=high,stream-format=byte-stream" ! h264parse ! queue ! fakesink`

`ximagesrc` outputs a BGRx format `video/x-raw` stream.

The same pipeline works totally fine in an AMD Vega Frontier Edition in the same environment. The GStreamer va plugin maintainer (He Junyan, Intel @HeJunyan) requested to post the issue here, as this is seemingly an Arc driver issue.

Backtrace:

[arc_segfault.txt](https://github.com/intel/media-driver/files/15287381/arc_segfault.txt)

[arc_segfault2.txt](https://github.com/intel/media-driver/files/15287397/arc_segfault2.txt)

[arc_bus.txt](https://github.com/intel/media-driver/files/15287388/arc_bus.txt)

Either the segfault or the bus error happens about 80-90% of the time (the parameters of the encoder and postproc doesn't matter), and even if the encoder is started, it is stuck at around 30 fps (the AMD encoder achieves full 60 fps). Similar issues existed back in Ubuntu 22.04 with Intel's repository but was expected to be fixed in Ubuntu 24.04 and kernel module 6.8.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

Basic postproc and encoding is very unstable on the Arc A750 GPU with Linux, thus usage for transcoding is very unreliable.

### Debug Information

1. Default in Ubuntu 24.04 (Kernel: 6.8.0-31-generic, intel-media-va-driver-non-free 24.1.0+ds1-1, libva 2.20.0), GStreamer 1.24.2
2. `by-path  card0  card1  renderD128  renderD129` & `pci-0000:07:00.0-card  pci-0000:07:00.0-render  pci-0000:0e:00.0-card  pci-0000:0e:00.0-render` (by-path)
3. 
`07:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A750] [8086:56a1] (rev 08)`
`0e:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XTX [Radeon Vega Frontier Edition] [1002:6863]`
4. 
```
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP
```
5.
[libva_trace.zip](https://github.com/intel/media-driver/files/15287419/libva_trace.zip)
6.
No Hanging

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-05-13:
unaligned size when copy vaImage data to surface

intel-mediadev, 2024-05-13:
Auto Created VSMGWL-73515 for further analysis.

fenhu, 2024-05-20:
@ehfd 
 I see your comments" Similar issues existed back in Ubuntu 22.04 with Intel's repository but was expected to be fixed in Ubuntu 24.04 and kernel module 6.8."  

Do you mean it is Kernel espcail issue?  

Thanks
Molly

ehfd, 2024-05-20:
@fenhu Not sure.

Check: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3331

XinfengZhang, 2024-05-22:
@fenhu , gdb --args gst-launch-1.0 ximagesrc show-pointer=0 remote=1 blocksize=16384 use-damage=0 ! "video/x-raw,framerate=60/1" ! vapostproc scale-method=fast ! "video/x-raw(memory:VAMemory),format=NV12" ! vah264lpenc aud=false b-frames=0 cpb-size=67 dct8x8=false key-int-max=1024 mbbrc=enabled num-slices=4 rate-control=cbr target-usage=4 bitrate=4000 ! "video/x-h264,profile=high,stream-format=byte-stream" ! h264parse ! queue ! fakesink

ehfd, 2024-05-22:
`gdb --args gst-launch-1.0 ximagesrc show-pointer=0 remote=1 blocksize=16384 use-damage=0 ! "video/x-raw,framerate=60/1" ! vapostproc scale-method=fast ! "video/x-raw(memory:VAMemory),format=NV12" ! vah264lpenc aud=false b-frames=0 cpb-size=67 dct8x8=false key-int-max=1024 mbbrc=enabled num-slices=4 rate-control=cbr target-usage=4 bitrate=4000 ! "video/x-h264,profile=high,stream-format=byte-stream" ! h264parse ! queue ! fakesink`

The funny thing here is that when gdb is used the SIGSEGV occurs less frequently.

Bin-CI, 2024-07-03:
Filter the history test data about upstream for 2 month refer media part on DG2 platform, couldn't find the same issue refer vah264lpenc

ehfd, 2024-08-08:
The issue should be mainly vapostproc, not vah264lpenc.

dbasabe, 2024-08-13:
Same problem here. 

Processor: 12th Gen Intel(R) Core(TM) i7-12700
GPU: Intel Corporation AlderLake-S GT1 (rev 0c)
OS: Ubuntu 22.04 (Kernel:6.5.0-45-generic, i915 driver, intel-media-va-driver-non-free 22.3.1+ds1-1ubuntu0.1 , libva 2.14.0-1) GStreamer 1.24.5

With:
_gst-launch-1.0 ximagesrc show-pointer=0 remote=1 blocksize=16384 use-damage=0 ! "video/x-raw,framerate=60/1" ! vapostproc scale-method=fast ! "video/x-raw(memory:VAMemory),format=NV12" ! vah264lpenc aud=false b-frames=0 cpb-size=67 dct8x8=false key-int-max=1024 mbbrc=enabled num-slices=4 rate-control=cbr target-usage=4 bitrate=4000 ! "video/x-h264,profile=high,stream-format=byte-stream" ! h264parse ! rtph264pay config-interval=1 ! queue ! udpsink host=0.0.0.0 port=8090
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'vah264lpenc0': gst.va.display.handle=context, gst-display=(GstObject)"\(GstVaDisplayDrm\)\ vadisplaydrm3", description=(string)"Intel\(R\)\ Gen\ Graphics", path=(string)/dev/dri/renderD128;
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV_

Sometimes it works and other times it randomly gives a SIGSEGV.

On the other hand, if I run it as @ehfd says using videconvert, it always works perfectly.

_gst-launch-1.0 ximagesrc show-pointer=0 remote=1 blocksize=16384 use-damage=0 ! "video/x-raw,framerate=60/1" ! videoconvert ! vah264lpenc aud=false b-frames=0 cpb-size=67 dct8x8=false key-int-max=1024 mbbrc=enabled num-slices=4 rate-control=cbr target-usage=4 bitrate=4000 ! "video/x-h264,profile=high,stream-format=byte-stream" ! h264parse ! rtph264pay config-interval=1 ! queue ! udpsink host=0.0.0.0 port=8090
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'vah264lpenc0': gst.va.display.handle=context, gst-display=(GstObject)"\(GstVaDisplayDrm\)\ vadisplaydrm2", description=(string)"Intel\(R\)\ Gen\ Graphics", path=(string)/dev/dri/renderD128;
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
Redistribute latency...
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:05.940566812
Setting pipeline to NULL ...
Freeing pipeline ..._

Effieyu0, 2024-09-09:
The issue is fixed on 722793d2eb507755affcacb151189e1bd052ce4c, 
Could you please to verify it @ehfd @dbasabe 
Thanks.

pengwan1, 2024-09-09:
pr: https://github.com/intel-innersource/drivers.gpu.unified/pull/178968

ehfd, 2024-09-12:
I think the fix works, although it was nothing long-running. Thank you.

Effieyu0, 2024-09-12:
@ehfd Thanks for your verifying.
Then I will close this issue.

closed 2024-09-12 by Effieyu0