intel/media-driver #1802: [Bug]: vainfo shows iHD_drv_video init failed

menggui1993, 2024-05-20:
### Is it regression? Good in old configuration?

None

### What happened?

I have a i3-1215u processor. I build libva, gmmlib and media-driver with source.

`lspci | grep VGA`
00:02.0 VGA compatible controller: Intel Corporation Device 46b3 (rev 0c)
**libva version**: 2.21.0
**gmmlib version**: 22.3.18
**media-driver version**: 24.1.5

**vainfo** shows
Trying display: wayland
error: XDG_RUNTIME_DIR not set in the environment.
Trying display: x11
error: can't connect to X server!
Trying display: drm
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-05-21:
What is your kernel version?
And could you install debug version media that will show critical error log in media driver.

menggui1993, 2024-05-21:
@Jexu 
kernel version: 5.15.49-rt47-intel-seer-rt

vainfo with debug infos:
Trying display: wayland
error: XDG_RUNTIME_DIR not set in the environment.
Trying display: x11
error: can't connect to X server!
Trying display: drm
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
[MOS]:  CRITICAL - mos_bufmgr_query_sys_engines:4741: Failed to query engines count.

[MOS]:  CRITICAL - HWInfo_GetGfxInfo:174: Failed to Init Gt System Info

[MOS]:  CRITICAL - Init:132: Fatal error - unsuccesfull Sku/Wa/GtSystemInfo initialization
[MOS]:  CRITICAL - CreateOsDeviceContext:128: MOS returned error, eStatus = 0x1b
[LIBVA]:CRITICAL - DdiMedia_InitMediaContext:1803: Unable to create MOS device context.
[LIBVA]:CRITICAL - DdiMedia__Initialize:1706: Failed to init media context
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

Jexu, 2024-05-21:
Pls check platform requirement, ADL-S/ADL-P: starting from kernel 5.17, try again by upgrading kernel?

menggui1993, 2024-05-22:
Solved, thanks.

closed 2024-05-22 by menggui1993