intel/media-driver #1807: [Bug]: musl build fails

zlice, 2024-05-29:
### Is it regression? Good in old configuration?

None

### What happened?

<details>
<summary>musl build error</summary>

[236/1359] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/media_interface/media_interfaces_m12_tgllp/media_interfaces_g12_tgllp.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/media_interface/media_interfaces_m12_tgllp/media_interfaces_g12_tgllp.cpp.o 
/usr/bin/g++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DIGFX_ARL_SUPPORTED -DIGFX_DG2_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_MTL_SUPPORTED -DIGFX_XEHP_SDV_CMFCPATCH_SUPPORTED -DIGFX_XEHP_SDV_SUPPORTED -DIGFX_XE_HPG_CMFCPATCH_SUPPORTED -DIGFX_XE_HPG_SUPPORTED -DMEDIA_VERSION=\"24.1.5\" -DMEDIA_VERSION_DETAILS=\"\" -DVEBOX_AUTO_DENOISE_SUPPORTED=0 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_FULL_OPEN_SOURCE -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MANUAL_SOFTLET_ -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_RELEASE -D_VP8_DECODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/builddir/intel-media-driver-24.1.5/media_softlet/linux/common/os/i915/include -I/builddir/intel-media-driver-24.1.5/media_softlet/linux/common/os/i915/include/uapi -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/os -I/builddir/intel-media-driver-24.1.5/media_driver/linux/common/os -I/builddir/intel-media-driver-24.1.5/media_common/agnostic/common/os/user_setting -I/builddir/intel-media-driver-24.1.5/media_common/agnostic/common/os -I/builddir/intel-media-driver-24.1.5/media_common/linux/common/os -I/builddir/intel-media-driver-24.1.5/media_softlet/agnostic/common/os/share -I/builddir/intel-media-driver-24.1.5/media_softlet/agnostic/common/os -I/builddir/intel-media-driver-24.1.5/media_softlet/linux/common/os/osservice -I/builddir/intel-media-driver-24.1.5/media_softlet/linux/common/os -I/builddir/intel-media-driver-24.1.5/media_common/linux/common/cp/os -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/cm -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/media_interfaces -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/shared/mediacopy -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/common/shared -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8/cm -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8_bdw/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen8_bdw/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9/cm -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9_bxt/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9_skl/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9_glk/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen9_kbl/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen11/cm -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen11/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen11/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen11/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen11_jsl_ehl/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen12/cm -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen12/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/gen12/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/g12/g12_base/hw/render -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_M/Xe_XPM/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_M/Xe_XPM/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_M/Xe_HPM/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_M/Xe_HPM/hw -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_R/Xe_HP_base/hw/render -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_R/Xe_HP_base/hw/blt -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_R/Xe_HP/hw/render -I/builddir/intel-media-driver-24.1.5/media_driver/agnostic/Xe_R/Xe_HPG/hw/render -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/common/shared/media_sfc_interface -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/common/shared/scalability -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/m12/m12_0/shared/mediacopy -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/g12/g12_base/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/g12/g12_0/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/g12/g12_1/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_R/Xe_HP_Base/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_R/Xe_HP/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_R/Xe_HPG/renderhal -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_M/Xe_XPM_base/shared/mediacopy -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_M/Xe_XPM_base/hw -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_M/Xe_HPM/shared/mediacopy -I/builddir/intel-media-driver-24.1.5/media_driver/media_softlet/agnostic/Xe_M/Xe_HPM/hw/vdbox -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m8_bdw -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m9_bxt -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m9_skl -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m9_cfl -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m9_glk -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m9_kbl -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m11_icllp -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m11_jsl_ehl -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_tgllp -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_dg1 -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_rkl -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_adls -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_adlp -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_m12_adln -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_xehp_sdv -I/builddir/intel-media-driver-24.1.5/media_driver/media_interface/media_interfaces_dg2 -I/builddir/intel-media-driver-24.1.5/media_common/agnostic/common/heap_manager -I/builddir/intel-media-drive [... cut ...]
[238/1359] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/media_interface/media_interfaces_dg2/media_interfaces_dg2.cpp.o
ninja: build stopped: subcommand failed.
=> ERROR: intel-media-driver-24.1.5_1: do_build: '${make_cmd} ${makejobs} ${make_build_args} ${make_build_target}' exited with 1
=> ERROR:   in do_build() at common/build-style/cmake.sh:95

</details>

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Build system. This seems like a similar issue https://github.com/intel/media-driver/issues/1639

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-05-31:
Auto Created VSMGWL-73973 for further analysis.

MicroYY, 2024-06-05:
@XinfengZhang Can we simply ignore these warnings by `-Wno-overloaded-virtual`? Otherwise we need a huge PR to totally fix virtual function being hidden.

zlice, 2024-06-05:
adding to CPPFLAGS / CXXFLAGS does fix the build for me

PR #1812 (2024-06-21): [Media Common] [VP] Disable overloaded virtual being hidden warning
Fix: #1807
commit 632f240a (2024-06-21): [Media Common] [VP] Disable overloaded virtual being hidden warning

Fix: #1807
closed 2024-06-21 by intel-mediadev