intel/media-driver #1809: [Bug]: VAAPI in Container with WSL2 Windows not working

DeltaTango69, 2024-05-31:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

[image]

docker-compose.yml

services:
  frigate:
    container_name: frigate
    privileged: true # this may not be necessary for all setups
    restart: unless-stopped
    image: ghcr.io/blakeblackshear/frigate:stable
    shm_size: "80mb" # update for your cameras based on calculation above
    devices:
      #- /dev/bus/usb:/dev/bus/usb  # Passes the USB Coral, needs to be modified for other versions
      #- /dev/apex_0:/dev/apex_0    # Passes a PCIe Coral, follow driver instructions here https://coral.ai/docs/m2/get-started/#2a-on-linux
      #- /dev/video11:/dev/video11  # For Raspberry Pi 4B
      - /dev/dri/renderD128:/dev/dri/renderD128 # For intel hwaccel, needs to be updated for your hardware
      - /dev/dri/card0:/dev/dri/card0
      - /dev/dri/:/dev/dri/
      - /dev/dxg/:/dev/dxg
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./config:/config
      - ./media:/media/frigate
      - type: tmpfs # Optional: 1GB of memory, reduces SSD/SD Card wear
        target: /tmp/cache
        tmpfs:
          size: 1000000000
    ports:
      - "5000:5000"
      - "8554:8554" # RTSP feeds
      - "8555:8555/tcp" # WebRTC over tcp
      - "8555:8555/udp" # WebRTC over udp
    environment:
      - TZ=Europe/Berlin
      - LIBVA_DRIVER_NAME=d3d12
      - LD_LIBRARY_PATH=/usr/lib/wsl/lib
      #- FRIGATE_RTSP_PASSWORD: "password"

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

# vainfo
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/vgem_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
# apt list mesa-va-drivers -a
Listing... Done
mesa-va-drivers/now 22.3.6-1+deb12u1 amd64 [installed,local]
mesa-va-drivers/oldstable 20.3.5-1 amd64

# vainfo --display drm --device /dev/dri/card0
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/vgem_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
# ^C
# ls /dev/dri
card0  renderD128
# ^C
# lspci -nn |grep -Ei 'VGA|DISPLAY'
/bin/sh: 12: lspci: not found
#

### Do you want to contribute a patch to fix the issue?

None

iori2333, 2024-10-15:
I found a [thread](https://community.intel.com/t5/Media-Intel-Video-Processing/Quicksync-support-for-WSL2/m-p/1312085) which said QSV wouldn't be supported for WSL2.

eric-gitta-moore, 2025-08-10:
The MSFT blog mentioned [D3D12 GPU Video acceleration in the Windows Subsystem for Linux now available!](https://devblogs.microsoft.com/commandline/d3d12-gpu-video-acceleration-in-the-windows-subsystem-for-linux-now-available/)

>To extend the types of workloads that we can accelerate with the GPU in WSLg, we also recently added support for GPU Video Acceleration by building on top of the existing [Mesa 3D](https://mesa3d.org/) D3D12 backend and integrating the [VAAPI](https://github.com/intel/libva) mesa frontend. Several linux media apps use the VAAPI interface to access hardware video acceleration when available, and this can now be leveraged in WSLg.