intel/media-driver #1810: [Bug]: Unable to use EncTools on DG1

iori2333, 2024-06-03:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

1. I'm using DG1 on Debian 12 with ffmpeg 7.0.1
2. Media Driver version: 24.1.5
3. Trigger EncTools (which provides software-based lookahead)
4. According to [Media Delivery Doc](https://github.com/intel/media-delivery/blob/master/doc/quality.rst), DG1 should support enabling EncTools using `-look_ahead_depth $lad`, yet the driver returns error: 
```
[hevc_qsv @ 0x563cc7cffc00] Error initializing the encoder: invalid video parameters (-15)
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
```

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

Transcoding command (modified from [Media Delivery scripts](https://github.com/intel/media-delivery/blob/master/scripts/ffmpeg-qsv-HEVC-1080p.sh)):
```bash
ffmpeg-qsv -v verbose -y -hwaccel qsv -qsv_device /dev/dri/renderD128 -extra_hw_frames 40 -c:v h264_qsv -i $stream \
        -c:v hevc_qsv -preset $preset \
        -vframes 6000 -profile:v main -async_depth $async \
        -b:v $bitrate -maxrate $((bitrate * 2)) -bufsize $((bitrate * 4)) \
        -rc_init_occupancy $((bitrate * 2)) -low_power true \
        -look_ahead_depth 40 -extbrc 1 -b_strategy 1 -bf 7 -refs 4 -g 300 \
        -idr_interval begin_only -strict -1 -fps_mode passthrough $output 2>&1 | tee transcode.log
```

Full Log:
```
[AVHWDeviceContext @ 0x563cc723dc40] libva: VA-API version 1.21.0
[AVHWDeviceContext @ 0x563cc723dc40] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x563cc723dc40] libva: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x563cc723dc40] libva: Found init function __vaDriverInit_1_20
[AVHWDeviceContext @ 0x563cc723dc40] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x563cc723dc40] Initialised VAAPI connection: version 1.21
[AVHWDeviceContext @ 0x563cc723dc40] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.1.5 ().
[AVHWDeviceContext @ 0x563cc723dc40] Driver not found in known nonstandard list, using standard behaviour.
[AVHWDeviceContext @ 0x563cc723ce00] Use Intel(R) oneVPL to create MFX session, API version is 2.11, the required implementation version is 1.3
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x563cc723ce00] Initialize MFX session: implementation version is 2.10
[h264 @ 0x563cc7391180] Reinit context to 1920x1088, pix_fmt: yuv420p
[vist#0:0/h264 @ 0x563cc816d840] WARNING: defaulting hwaccel_output_format to qsv for compatibility with old commandlines. This behaviour is DEPRECATED and will be removed in the future. Please explicitly set "-hwaccel_output_format qsv".
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'source.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.30.100
  Duration: 02:48:40.98, start: 0.000000, bitrate: 5877 kb/s
  Stream #0:0[0x1](und): Video: h264 (Main), 1 reference frame (avc1 / 0x31637661), yuv420p(progressive, left), 1920x1080 [SAR 1:1 DAR 16:9], 5674 kb/s, 29.97 fps, 29.97 tbr, 30k tbn (default)
      Metadata:
        handler_name    : VideoHandler
        vendor_id       : [0][0][0][0]
  Stream #0:1[0x2](und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 192 kb/s (default)
      Metadata:
        handler_name    : SoundHandler
        vendor_id       : [0][0][0][0]
[out#0/matroska @ 0x563cc8168280] No explicit maps, mapping streams automatically...
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Created video stream from input stream 0:0
[aost#0:1/libvorbis @ 0x563cc7b25540] Created audio stream from input stream 0:1
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_qsv) -> hevc (hevc_qsv))
  Stream #0:1 -> #0:1 (aac (native) -> vorbis (libvorbis))
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Starting thread...
[aost#0:1/libvorbis @ 0x563cc7b25540] Starting thread...
[vf#0:0 @ 0x563cc7428600] Starting thread...
[af#0:1 @ 0x563cc74284c0] Starting thread...
[vist#0:0/h264 @ 0x563cc816d840] [dec:h264_qsv @ 0x563cc8149f80] Starting thread...
[aist#0:1/aac @ 0x563cc7b2ae00] [dec:aac @ 0x563cc8148380] Starting thread...
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400] Starting thread...
Press [q] to stop, [?] for help
[AVHWDeviceContext @ 0x7fec9000a1c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.1.5 ().
[AVHWDeviceContext @ 0x7fec9000a1c0] Driver not found in known nonstandard list, using standard behaviour.
[h264_qsv @ 0x563cc814aac0] Decoder: output is video memory surface
[h264_qsv @ 0x563cc814aac0] Use Intel(R) oneVPL to create MFX session with the specified MFX loader
[graph_1_in_0:1 @ 0x7fec8c003740] tb:1/48000 samplefmt:fltp samplerate:48000 chlayout:stereo
[AVHWDeviceContext @ 0x7fec900103c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.1.5 ().
[AVHWDeviceContext @ 0x7fec900103c0] Driver not found in known nonstandard list, using standard behaviour.
[h264_qsv @ 0x563cc814aac0] Decoder: output is video memory surface
[h264_qsv @ 0x563cc814aac0] Use Intel(R) oneVPL to create MFX session with the specified MFX loader
[graph 0 input from stream 0:0 @ 0x7fec94003100] w:1920 h:1080 pixfmt:qsv tb:1/30000 fr:30000/1001 sar:1/1 csp:unknown range:tv
[graph 0 input from stream 0:0 @ 0x7fec94003100] video frame properties congruent with link at pts_time: 0
[hevc_qsv @ 0x563cc7cffc00] Using input frames context (format qsv) with hevc_qsv encoder.
[hevc_qsv @ 0x563cc7cffc00] Encoder: input is video memory surface
[hevc_qsv @ 0x563cc7cffc00] Use Intel(R) oneVPL to create MFX session with the specified MFX loader
[hevc_qsv @ 0x563cc7cffc00] Using the variable bitrate (VBR) ratecontrol method
[hevc_qsv @ 0x563cc7cffc00] Error initializing the encoder: invalid video parameters (-15)
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
[vf#0:0 @ 0x563cc7428600] Error sending frames to consumers: Invalid argument
[vf#0:0 @ 0x563cc7428600] Task finished with error code: -22 (Invalid argument)
[vf#0:0 @ 0x563cc7428600] Terminating thread with return code -22 (Invalid argument)
[vist#0:0/h264 @ 0x563cc816d840] [dec:h264_qsv @ 0x563cc8149f80] Decoder returned EOF, finishing
[vist#0:0/h264 @ 0x563cc816d840] [dec:h264_qsv @ 0x563cc8149f80] Terminating thread with return code 0 (success)
[vist#0:0/h264 @ 0x563cc816d840] All consumers of this stream are done
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400] Terminating thread with return code 0 (success)
[aist#0:1/aac @ 0x563cc7b2ae00] [dec:aac @ 0x563cc8148380] Decoder thread received EOF packet
[aist#0:1/aac @ 0x563cc7b2ae00] [dec:aac @ 0x563cc8148380] Decoder returned EOF, finishing
[aist#0:1/aac @ 0x563cc7b2ae00] [dec:aac @ 0x563cc8148380] Terminating thread with return code 0 (success)
[af#0:1 @ 0x563cc74284c0] Filtergraph returned EOF, finishing
[af#0:1 @ 0x563cc74284c0] All consumers returned EOF
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Encoder thread received EOF
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Could not open encoder before EOF
[aost#0:1/libvorbis @ 0x563cc7b25540] Encoder thread received EOF
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Task finished with error code: -22 (Invalid argument)
[vost#0:0/hevc_qsv @ 0x563cc7b4e080] Terminating thread with return code -22 (Invalid argument)
[aost#0:1/libvorbis @ 0x563cc7b25540] Terminating thread with return code 0 (success)
[af#0:1 @ 0x563cc74284c0] Terminating thread with return code 0 (success)
[out#0/matroska @ 0x563cc8168280] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0KiB time=N/A bitrate=N/A speed=N/A
[AVIOContext @ 0x563cc8155300] Statistics: 0 bytes written, 0 seeks, 0 writeouts
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400] Input file #0 (source.mp4):
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400]   Input stream #0:0 (video): 26 packets read (24612 bytes); 10 frames decoded; 0 decode errors;
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400]   Input stream #0:1 (audio): 11166 packets read (5605984 bytes); 11164 frames decoded; 0 decode errors (11431936 samples);
[in#0/mov,mp4,m4a,3gp,3g2,mj2 @ 0x563cc7d32400]   Total: 11192 packets (5630596 bytes) demuxed
[AVIOContext @ 0x563cc7b2b340] Statistics: 183620120 bytes read, 2 seeks
Conversion failed!
frame=    0 fps=0.0 q=0.0 Lsize=       0KiB time=N/A bitrate=N/A speed=N/A
```

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-06-03:
Auto Created VSMGWL-74004 for further analysis.

iori2333, 2024-06-05:
> @iori2333 : can you, please, help clarify your environment? specifically:
> 
>     * Which kernel are you using? `uname -a`
> 
>     * How did you install media stack? did you use any pre-built packages or built from sources?
> 
>     * If you built from sources, which tags and configure command lines you've used for:
>       
>       * media driver?
>       * mediasdk? https://github.com/Intel-Media-SDK/MediaSDK
>       * vpl? https://github.com/intel/libvpl
>       * vpl runtime? https://github.com/intel/vpl-gpu-rt
> 
>     * How did you configure and built ffmpeg? /in particular, you've used `--enable-libmfx` or `--enable-libvpl`/

- I'm using Linux 6.5 on Debian. I also tried Linux LTS 6.1, which doesn't help either.
- Upstream kernel doesn't contain drivers for DG1, so I built and installed [OOT Drivers](https://github.com/intel-gpu/intel-gpu-i915-backports) and [Media Driver](https://github.com/Intel-Media-SDK/MediaSDK).
- I use Docker image from [linuxserver](https://hub.docker.com/r/linuxserver/ffmpeg) to use ffmpeg, which provides VPL runtime.

Command Lines:
```bash
# uname -a
Linux myhost 6.5.0-0.deb12.4-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.10-1~bpo12+1 (2023-11-23) x86_64 GNU/Linux

# Backport driver (https://github.com/intel-gpu/intel-gpu-i915-backports/blob/backport/main/docs/README_vanilla.md)
make i915dkmsdeb-pkg # use release [I915_24WW22.3_803.58_23.10.49_231129.48](https://github.com/intel-gpu/intel-gpu-i915-backports/releases/tag/I915_24WW22.3_803.58_23.10.49_231129.48)

# Media driver
cmake .. -DENABLE_PRODUCTION_KMD=ON # according to README for DG1

# ffmpeg flags
ffmpeg version 7.0 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.2.0-23ubuntu4)
  configuration: --disable-debug --disable-doc --disable-ffplay --enable-alsa --enable-cuvid --enable-ffprobe --enable-gpl --enable-libaom --enable-libass --enable-libfdk_aac --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libharfbuzz --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libplacebo --enable-librav1e --enable-libshaderc --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nonfree --enable-nvdec --enable-nvenc --enable-cuda-llvm --enable-opencl --enable-openssl --enable-stripping --enable-vaapi --enable-vdpau --enable-version3 --enable-vulkan
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
Universal media converter
```

iori2333, 2024-06-05:
I copied command lines for building VPL and VPL runtime from their [Dockerfile](https://github.com/linuxserver/docker-ffmpeg/blob/master/Dockerfile)
```bash
RUN \
  echo "**** grabbing libvpl ****" && \
  mkdir -p /tmp/libvpl && \
  curl -Lf \
    https://github.com/intel/libvpl/archive/refs/tags/v${LIBVPL}.tar.gz | \
    tar -zx --strip-components=1 -C /tmp/libvpl
RUN \
  echo "**** compiling libvpl ****" && \
  mkdir -p /tmp/libvpl/build && \
  cd /tmp/libvpl/build && \
  cmake .. && \ 
  cmake --build . --config Release && \
  cmake --build . --config Release --target install && \
  strip -d /usr/local/lib/libvpl.so
RUN \
  echo "**** grabbing vpl-gpu-rt ****" && \
  mkdir -p /tmp/vpl-gpu-rt && \
  curl -Lf \
    https://github.com/intel/vpl-gpu-rt/archive/refs/tags/intel-onevpl-${VPLGPURT}.tar.gz | \
    tar -zx --strip-components=1 -C /tmp/vpl-gpu-rt
RUN \
  echo "**** compiling vpl-gpu-rt ****" && \
  mkdir -p /tmp/vpl-gpu-rt/build && \
  cd /tmp/vpl-gpu-rt/build && \
  cmake \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=/usr/local/lib \
    .. && \ 
  make && \
  make install && \
  strip -d /usr/local/lib/libmfx-gen.so
```

dvrogozh, 2024-06-11:
@iori2333 : unfortunately, I could not reproduce your issue on my environment with DG1. Could you, please, try to build updated media-delivery from this branch: https://github.com/dvrogozh/media-delivery/tree/for-1810. I update user space components to be closer to what you've tried. Please, try to build as follows and run on the same kernel as you already configured. I suggest the following experiments:
```
docker build -f docker/ubuntu20.04/selfbuild-prodkmd/Dockerfile -t mds .
# the following assumes that DG1 is at renderD128 - change accordingly as needed:
docker run -it --rm --device=/dev/dri/renderD128 --group-add=109 --cap-add SYS_ADMIN mds

ffmpeg-qsv-HEVC-1080p.sh 1

ffmpeg -v verbose -y -hwaccel qsv -qsv_device /dev/dri/renderD128 -extra_hw_frames 40 -c:v hevc_qsv -i bbb_sunflower_1080p_60fps_4Mbps_38kframes.h265     -c:v hevc_qsv -preset veryfast     -vframes 6000 -profile:v main -async_depth 1     -b:v 3000000 -maxrate 6000000 -bufsize 12000000     -rc_init_occupancy 6000000 -low_power true     -look_ahead_depth 40 -extbrc 1 -b_strategy 1 -bf 7 -refs 4 -g 300     -idr_interval begin_only -strict -1 -fps_mode passthrough out.h265

wget https://repositories.intel.com/media/bbb_sunflower_1080p_60fps_4Mbps_38kframes.h264

ffmpeg -v verbose -y -hwaccel qsv -qsv_device /dev/dri/renderD128 -extra_hw_frames 40 -c:v h264_qsv -i bbb_sunflower_1080p_60fps_4Mbps_38kframes.h264     -c:v hevc_qsv -preset veryfast     -vframes 6000 -profile:v main -async_depth 1     -b:v 3000000 -maxrate 6000000 -bufsize 12000000 out.h265
```

If these will work, try your own stream which you tried before to check whether that's stream specific.

iori2333, 2024-06-12:
@dvrogozh Thanks, I will have a try

iori2333, 2024-06-22:
Thanks for @dvrogozh's script, now EncTools works fine. I will try rebuilding VPL related libs.

closed 2024-06-22 by iori2333