intel/media-driver #1819: JPEG decoding doesn't work

RankoR, 2024-06-20:
I'm running Arch on i9-14900K. `libva` and `libva-utils` are installed.

```
loadjpeg ~/Downloads/jpeg/random_noise_1.jpg                                                                                                                                                                                                                                   
WARNING:Height need to be a multiple of 16 (current height is 1080)
Decoding JPEG image 1920x1080...
Trying display: wayland
loadjpeg: ../libva-utils-2.21.0/decode/tinyjpeg.c:612: tinyjpeg_decode: Assertion `0' failed.
[1]    269509 IOT instruction (core dumped)  loadjpeg ~/Downloads/jpeg/random_noise_1.jpg
```

When I'm running this C++ code:

```
const int fd = open("/dev/dri/renderD128", O_RDWR);
    vaDisplay = vaGetDisplayDRM(fd);
    int major, minor;
    VAStatus status = vaInitialize(vaDisplay, &major, &minor);
    if (status != VA_STATUS_SUCCESS) {
        throw std::runtime_error("Failed to initialize VA display");
    }

    VAEntrypoint entrypoints[5];
    int num_entrypoints, vld_entrypoint;
    status = vaQueryConfigEntrypoints(vaDisplay, VAProfileJPEGBaseline, entrypoints,
                                      &num_entrypoints);

    if (status != VA_STATUS_SUCCESS) {
        throw std::runtime_error("Failed to query config entrypoints");
    }

    for (vld_entrypoint = 0; vld_entrypoint < num_entrypoints; vld_entrypoint++) {
        if (entrypoints[vld_entrypoint] == VAEntrypointVLD)
            break;
    }

    if (vld_entrypoint == num_entrypoints) {
        throw std::runtime_error("Failed to find VLD entry point");
    }

    VASurfaceAttrib forcc;
    forcc.type = VASurfaceAttribPixelFormat;
    forcc.flags = VA_SURFACE_ATTRIB_SETTABLE;
    forcc.value.type = VAGenericValueTypeInteger;

    // Create VA configuration and context
    VAConfigAttrib configAttrib;
    configAttrib.type = VAConfigAttribRTFormat;
    VAProfile profile;
    VAEntrypoint entrypoint;

    status = vaGetConfigAttributes(vaDisplay, VAProfileJPEGBaseline, VAEntrypointVLD,
                          &configAttrib, 1);

    if (status != VA_STATUS_SUCCESS) {
        throw std::runtime_error("Failed to get config attributes");
    }
```

It returns `VA_STATUS_ERROR_UNSUPPORTED_PROFILE` from `vaGetConfigAttributes`.

Or if I call 

```
status = vaCreateConfig(vaDisplay, VAProfileJPEGBaseline, VAEntrypointVLD, &configAttrib, 1, &vaConfig);
```

I'm getting the same result.

But, however, there is not error when I change `VAProfileJPEGBaseline` to `VAProfileH264Main` or `VAProfileVP9Profile0` for example.

So, what's the problem and how to decode JPEG?

Jexu, 2024-06-27:
How about vainfo? It will list all supported profiles.

RankoR, 2024-06-27:
```
Trying display: wayland
vainfo: VA-API version: 1.21 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
```
Hm, not many of them. My CPU should support more, right?

Jexu, 2024-06-27:
Do you open intel driver? it looks NVDEC driver opened? And multi gpu exists on your machine?

RankoR, 2024-06-27:
@Jexu yep, posted wrong output originally. I have Nvidia GPU, and can't find out yet how to launch vainfo for the iGPU, it uses nvidia by default.

Jexu, 2024-06-27:
Try to open another gpu node under /dev/dri/renderD*

intel-mediadev, 2024-06-27:
Auto Created VSMGWL-74546 for further analysis.

Jexu, 2024-07-05:
> Try to open another gpu node under /dev/dri/renderD*

May I know if it solves your issue.

Jexu, 2024-07-18:
I will close this issue now since it is caused by opening wrong gpu, issue could be solved by opening intel gpu. Let me know if you still have any other concerns.

closed 2024-07-18 by Jexu