intel/media-driver #1820: [Bug]: gstreamer plugin vapostproc produces

z0Kng, 2024-06-20:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

None

### What happened?

Using this pipeline gstreamer

`gst-launch-1.0 videotestsrc ! video/x-raw,format=BGRx ! vapostproc ! "video/x-raw(memory:VAMemory),format=NV12" ! vah265lpenc ! h265parse ! matroskamux ! filesink location=test.mkv`

produces a tiled image on an intel flex 140. This bug also happens for BGRx, xRGB, xBGR and RGBx as input, while the formats with alpha channel work. Using msdkvpp instead of vapostproc works. Using vaapipostproc and vaapih265enc instead also.
[image]

I only could reproduce this bug on an Intel Flex GPU.

I have already opened an issue at gestreamer: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3634 and was asked to open another one here as it is suspected to be a driver problem.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

We are currently developing an opensource vdi solution using Flex vGPUs for graphics acceleration and video stream encoding.

### Debug Information

1. vainfo: VA-API version: 1.19 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.2.3 (968bb0173)
2. exists 
3. 
 ```
gputest02:~$ lspci -nn |grep  -Ei 'VGA|DISPLAY'
07:00.0 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.1 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.2 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.3 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.4 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.5 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
07:00.6 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.0 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.1 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.2 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.3 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.4 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.5 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0a:00.6 Display controller [0380]: Intel Corporation Device [8086:56c1] (rev 05)
0c:00.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 41)
43:00.0 Display controller [0380]: Adv
anced Micro Devices, Inc. [AMD/ATI] Arcturus GL-XL [Instinct MI100] [1002:738c] (rev 01)
```

4.[vainfo.log](https://github.com/user-attachments/files/15913100/vainfo.log)
5.[libva_trace.log](https://github.com/user-attachments/files/15913099/libva_trace.log)

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-06-26:
Auto Created VSMGWL-74530 for further analysis.

VincentCheungKokomo, 2024-07-04:
@z0Kng Could you share the reproduce environment to me for this issue

z0Kng, 2024-07-08:
I have more information,
My setup is:
- **Operating System:** Ubuntu 22.04.4 LTS x86_64 (Server)
- **GStreamer Version:** 1.25 (pulled from master)
- **GPU:** Intel® Data Center GPU Flex 140
- **VA-API version:** 1.22 (libva 2.20.0)
- **Media Driver version:** Intel iHD driver for Intel(R) Gen Graphics - 24.2.5 (12561f682)
- **intel-i915-dkms:** 1.23.10.49.231129.48+i77-1

Furthermore, the following error messages occur when using gst-inspect or using the plugins:

```
0:00:00.450089145 85446 0x55a544aeb950 ERROR               vamemory gstvaallocator.c:616:_va_create_surface_and_export_to_dmabuf: Different objects have different modifier

(gst-plugin-scanner:85446): GStreamer-CRITICAL **: 08:01:31.514: gst_caps_append: assertion 'GST_IS_CAPS (caps2)' failed

```
Also the pipeline crashes after some time:

```
gst-launch-1.0 videotestsrc ! video/x-raw,format=BGRx ! vapostproc ! "video/x-raw(memory:VAMemory),format=NV12" ! vah265lpenc ! h265parse ! matroskamux ! filesink location=test.mkv

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vah265lpenc0': gst.va.display.handle=context, gst-display=(GstObject)"\(GstVaDisplayDrm\)\ vadisplaydrm6", description=(string)"Intel\(R\)\ Gen\ Graphics", path=(string)/dev/dri/renderD128;
Redistribute latency...
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
Redistribute latency...
New clock: GstSystemClock
Bus error (core dumped)
```

MicroYY, 2024-08-14:
Fixed by 019772d. Close it

closed 2024-08-14 by MicroYY