intel/media-driver #1821: [Bug]: Transcoding failed with GPU HANG

Goodwu, 2024-06-22:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

1. Hardware is J3455
2. Immich docker in OMV8 in PVE container
3. Transcoding with HW decoding and encoding on
4. Transcoding failed with GPU HANG

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Transcoding failed

### Debug Information

/usr/lib/jellyfin-ffmpeg/vainfo
```
Trying display: drm
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.2.1 (0593864)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
```

error log from ffmpeg:
```
[Nest] 7  - 06/20/2024, 2:05:59 PM     LOG [Microservices:MediaService] Started encoding video 7de4b8c4-6638-4d00-bf1d-7b731e4e9bb8 {"inputOptions":["-hwaccel qsv","-hwaccel_output_format qsv","-async_depth 4","-threads 1"],"outputOptions":["-c:v h264_qsv","-c:a copy","-movflags faststart","-fps_mode passthrough","-map 0:0","-map 0:1","-bf 7","-refs 5","-g 256","-v verbose","-vf scale_qsv=720:-1:async_depth=4:mode=hq,hwmap=derive_device=opencl,tonemap_opencl=desat=0:format=nv12:matrix=bt709:primaries=bt709:range=pc:tonemap=hable:transfer=bt709,hwmap=derive_device=qsv:reverse=1,format=qsv","-preset 7","-global_quality:v 23","-maxrate 4000k","-bufsize 8000k"],"twoPass":false}
[Nest] 7  - 06/20/2024, 2:06:06 PM   ERROR [Microservices:MediaRepository] [AVHWDeviceContext @ 0x49a52032040] Initialised VAAPI connection: version 1.21
[AVHWDeviceContext @ 0x49a52032040] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 24.2.1 (0593864).
[... 374 lines cut ...]
[530927.644808] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:8ed1fff2, in ffmpeg [47795]
[530964.882985] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[530964.883088] i915 0000:00:02.0: [drm] ffmpeg[48202] context reset due to GPU hang
[530964.890020] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:8ed1fff2, in ffmpeg [48202]
[530983.441819] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[530983.441925] i915 0000:00:02.0: [drm] ffmpeg[48405] context reset due to GPU hang
[530983.452043] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:8ed1fff2, in ffmpeg [48405]
[531032.747598] i915 0000:00:02.0: [drm] Resetting rcs0 for CS error
[531032.747704] i915 0000:00:02.0: [drm] ffmpeg[48902] context reset due to GPU hang
[531032.755920] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:00280000, in ffmpeg [48902]
[531055.584224] i915 0000:00:02.0: [drm] Resetting rcs0 for CS error
[531055.584326] i915 0000:00:02.0: [drm] ffmpeg[49112] context reset due to GPU hang
[531055.602631] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:00280000, in ffmpeg [49112]
[531078.927740] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[531078.927849] i915 0000:00:02.0: [drm] ffmpeg[49382] context reset due to GPU hang
[531078.935235] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:e757fefe, in ffmpeg [49382]
[531113.717433] i915 0000:00:02.0: [drm] Resetting rcs0 for CS error
[531113.717537] i915 0000:00:02.0: [drm] ffmpeg[49734] context reset due to GPU hang
[531113.724791] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:00280000, in ffmpeg [49734]
[581042.737142] BTRFS info (device dm-11): device stats zeroed by btrfs (495467)
```

### Do you want to contribute a patch to fix the issue?

None

Goodwu, 2024-06-23:
[https://pan.baidu.com/s/1DR_2FTUGH5gGJX2luQqSOQ?pwd=v5pp](url)

Goodwu, 2024-06-23:
The transcoding hangs ffmpeg randomly.
This is the transcoding command line:
`/usr/bin/ffmpeg -hwaccel qsv -hwaccel_output_format qsv -async_depth 4 -threads 1 -i /media/photos/Camera Uploads/DCIM/Camera/VID_20240109_121531.mp4 -y -c:v h264_qsv -c:a copy -movflags faststart -fps_mode passthrough -map 0:0 -map 0:1 -bf 7 -refs 5 -g 256 -v verbose -vf scale_qsv=720:-1:async_depth=4:mode=hq,hwmap=derive_device=opencl,tonemap_opencl=desat=0:format=nv12:matrix=bt709:primaries=bt709:range=pc:tonemap=hable:transfer=bt709,hwmap=derive_device=qsv:reverse=1,format=qsv -preset 7 -global_quality:v 23 -maxrate 4000k -bufsize 8000k upload/encoded-video/0ef79aea-c9d6-4f20-b7d2-bc3a01408cb6/28/49/28491f52-ccf9-4073-aace-b7919fffd528.mp4`

XinfengZhang, 2024-06-24:
@xhaihao , looks the hang is in RCS, so , suppose it is caused by tonemap_opencl, it is not a media driver feature.  should be reported in ffmpeg or openCL?

Goodwu, 2024-06-28:
> @xhaihao , looks the hang is in RCS, so , suppose it is caused by tonemap_opencl, it is not a media driver feature. should be reported in ffmpeg or openCL?

Is it similar to #1456 ?

chao-camect, 2024-07-24:
I have the same issue. I suspect it's some change after `intel-media-23.2.4`.
It has been running fine for me for years, before I upgraded to a newer version.
I'll revert to `intel-media-23.2.4` and keep other deps the same see whether it'll be fine.

chao-camect, 2024-08-21:
Have been running well with `intel-media-23.2.4` on kernel `5.19.17` for almost a month.
I'll switch back to the latest release again to see whether I can reproduce it.