intel/media-driver #1825: [Bug]: Raptor lake 13500H crash

4e4o, 2024-06-30:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Does raptor lake H support planned? I have some crash in dmesg, intel-media-driver-24.1.5-1.fc40.x86_64 .
[dmesg_crash.txt](https://github.com/user-attachments/files/16045055/dmesg_crash.txt)

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-07-01:
I don't see anything unusual in dmesg, please give more descriptions of what crash it is, any crash log from media driver? How about vainfo?

4e4o, 2024-07-01:
> I don't see anything unusual in dmesg, please give more descriptions of what crash it is, any crash log from media driver? How about vainfo?

is i915 from intel-media driver? Should i post this issue to kernel repo? I can't see supported Raptor Lake H GPU in supprorted list: TGLx (TGL: Tiger Lake, RKL: Rocket Lake, ADL-S/P/N: Alder Lake, RPL-S/P: Raptor Lake). There is only S/P versions.  Also intel_gpu_top detects gpu as Alder_lake_p (Gen_12) - is it normal ? vainfo and intel_gpu_top output attached.
[vainfo.txt](https://github.com/user-attachments/files/16049062/vainfo.txt)
[intel_gpu_top.txt](https://github.com/user-attachments/files/16049064/intel_gpu_top.txt)

4e4o, 2024-07-01:
Here is full dmesg just in case.
[dmesg.txt](https://github.com/user-attachments/files/16049101/dmesg.txt)

Jexu, 2024-07-01:
13500H should be rpl-p, and your vainfo looks good since both media driver and i915 driver all initiaed successffully.

4e4o, 2024-07-01:
> 13500H should be rpl-p, and your vainfo looks good since both media driver and i915 driver all initiaed successffully.

Oh , sorry, this is only a warning:
WARNING: CPU: 0 PID: 468 at drivers/gpu/drm/i915/display/intel_bios.c:2678 intel_bios_init+0x1821/0x2770 [i915]  

Ok i will close this one. Thank you very much.

closed 2024-07-01 by 4e4o