intel/media-driver #1827: [Bug]: 32a5aa6528f8 build error on Ubuntu Noble(2404)

Bin-CI, 2024-07-09:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Build error on Ubuntu Noble(2404), Ubuntu Jammy(2204) without this error
Noble use GCC 13 version, Jammy use GCC 11 version

34429a3fad0c pass
32a5aa6528f8 error

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Bin-CI, 2024-07-09:
Build error information
```
18:03:30,877 INFO  - /opt/media/build/intel-media-driver/media_softlet/agnostic/common/shared/packet/media_render_cmd_packet.cpp: In member function ���MOS_STATUS RenderCmdPacket::PrepareComputeWalkerParams(KERNEL_WALKER_PARAMS, MHW_GPGPU_WALKER_PARAMS&)���:
18:03:30,877 INFO  - /opt/media/build/intel-media-driver/media_softlet/agnostic/common/shared/packet/media_render_cmd_packet.cpp:1267:39: error: storing the address of local variable ���params��� in ���*gpgpuWalker._MHW_GPGPU_WALKER_PARAMS::inlineDataParamBase��� [-Werror=dangling-pointer=]
18:03:30,877 INFO  -  1267 |     gpgpuWalker.inlineDataParamBase   = params.inlineDataParams;
18:03:30,877 INFO  -       |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
18:03:30,878 INFO  - /opt/media/build/intel-media-driver/media_softlet/agnostic/common/shared/packet/media_render_cmd_packet.cpp:1199:77: note: ���params��� declared here
18:03:30,878 INFO  -  1199 | MOS_STATUS RenderCmdPacket::PrepareComputeWalkerParams(KERNEL_WALKER_PARAMS params, MHW_GPGPU_WALKER_PARAMS &gpgpuWalker)
18:03:30,878 INFO  -       |                                                        ~~~~~~~~~~~~~~~~~~~~~^~~~~~
18:03:30,878 INFO  - /opt/media/build/intel-media-driver/media_softlet/agnostic/common/shared/packet/media_render_cmd_packet.cpp:1199:110: note: ���gpgpuWalker��� declared here
18:03:30,878 INFO  -  1199 | MOS_STATUS RenderCmdPacket::PrepareComputeWalkerParams(KERNEL_WALKER_PARAMS params, MHW_GPGPU_WALKER_PARAMS &gpgpuWalker)
18:03:30,878 INFO  -       |                                                                                     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
18:03:31,468 INFO  - cc1plus: all warnings being treated as errors
18:03:31,479 INFO  - make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_SOFTLET_COMMON.dir/build.make:118: media_driver/CMakeFiles/iHD_drv_video_SOFTLET_COMMON.dir/__/media_softlet/agnostic/common/shared/packet/media_render_cmd_packet.cpp.o] Error 1
18:03:31,479 INFO  - make[2]: Leaving directory '/opt/media/build/intel-media-driver/__cmake_build'
18:03:31,479 INFO  - make[1]: *** [CMakeFiles/Makefile2:261: media_driver/CMakeFiles/iHD_drv_video_SOFTLET_COMMON.dir/all] Error 2
```

Bin-CI, 2024-07-09:
Add option `-DCMAKE_CXX_COMPILER=g++-11` can fix the build error

uartie, 2024-07-10:
It’s broken on Fedora 40, too, which uses GCC 14.  And F40 does not appear to have GCC 11 package to fallback on.

MicroYY, 2024-07-12:
The fix has been merged in internal repo and will be ported to this repo soon.

Bin-CI, 2024-07-15:
Build fixed on Noble

closed 2024-07-15 by Bin-CI