intel/media-driver #1830: [Bug]: Fail to play 2048x1152 mpeg2 video with oneVPL and media-driver.

Shao-Feng, 2024-07-11:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

1 On Android, using default player "Gallery" to play a mpeg2 2K (2048x1152) video
2 With oneVPL and media-driver, 2k mpeg2 video should be played well.
3 In oneVPL https://github.com/intel/vpl-gpu-rt/blob/main/_studio/shared/src/libmfx_core_vaapi.cpp#L1631
the value for width and height returned from driver is 1920 for mpeg2, instead of 2048.
va_sts = vaGetConfigAttributes(*m_p_display_wrapper, req_profile,
                                   req_entrypoint,
                                   attr, sizeof(attr)/sizeof(*attr));
and then fail to pass oneVPL check for max width and height
on https://github.com/intel/vpl-gpu-rt/blob/main/_studio/shared/src/libmfx_core_vaapi.cpp#L1641
and https://github.com/intel/vpl-gpu-rt/blob/main/_studio/shared/src/libmfx_core_vaapi.cpp#L1642
4 the expected max width and height for mpeg2 is 2048

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

Shao-Feng, 2024-07-11:
Once removing the width and height checking from oneVPL. the 2k 2048x1152 mpeg2 video can be played well.

Jexu, 2024-07-11:
Miss setting correct max pic width and height for mpeg2 decoding in https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L2353

PR #1831 (2024-07-16): Add max resolution for MPEG2 on Gen12
Max resolution of MPEG2 is 2K.
assigning 2048x2048 for MPEG2 attributies

fixes: #1830
commit 213acc4b (2024-07-16): Add max resolution for MPEG2 on Gen12

Max resolution of MPEG2 is 2K.
assigning 2048x2048 for MPEG2 attributies

fixes: #1830
closed 2024-07-16 by intel-mediadev