intel/media-driver #1832: [Feature]: AV1 decode support in media-driver without any kernels built

frantisekz, 2024-07-12:
### What Feature?

AV1 decode is currently exposed in vainfo on supported HW, but the decoding itself fails if the media-driver is built without any kernels (neither free nor non-free), supposedly because the required filmgrain kernel for AV1 decode. 

Would it be possible to implement some fallback path so the driver makes the decoding at least partially on GPU on supported hardware even without any kernels built in?

I can add ffmpeg verbose log once I get to the HW that supports AV1.

Thanks!

### What's the usage scenario would be benifited?

Playback, Web browser

### Do you want to contribute a patch to develop this feature?

None

Sherry-Lin, 2024-07-15:
Thanks for your feedback. Media-driver default build supports full features. Could you provide more background why not choose default build?

frantisekz, 2024-07-15:
Hey @Sherry-Lin , thanks for the reply!

This is due to media-driver package in Fedora, where I, as a package maintainer, can't use prebuilt binaries due to patents and distribution policy, and rebuilding the kernels is broken as discussed in #1775 and #1356 .

intel-mediadev, 2024-07-15:
Auto Created VSMGWL-74924 for further analysis.

Jexu, 2024-07-17:
AV1 filmgrain kernel only exists on several platforms and it already moves to fix function from mtl without kernel anymore.
I think it maybe hard to get priorized so far unless having any customer special usage.

frantisekz, 2024-07-17:
Okay, thanks @Jexu , that's good to know, I am not able to test this as the latest HW I have available fails to decode the AV1 without kernels (Xe iGPU).