intel/media-driver #1833: [Bug]: `scale_vaapi` behaves differently than `zscale` (ffmpeg)

Anuskuss, 2024-07-13:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

None

### What happened?

I'm usually do HDR-to-SDR tonemapping using the following command:
```bash
ffmpeg -i hdr.mp4 -vf zscale=transfer=linear,tonemap=mobius,zscale=transfer=bt709,format=yuv420p thumb-%06.png
```
but unfortunately I can't using `zscale` so I've tried to use VAAPI instead:
```bash
ffmpeg -i hdr.mp4 -vf hwupload,scale_vaapi=format=p010,hwdownload,format=p010,tonemap=mobius,hwupload,scale_vaapi=format=nv12,hwdownload,format=nv12 thumb-%06.png
```
This creates some vastly different colors though. I don't think I'm doing something wrong (it's almost the same exact command) so I decided to open this issue.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Do you want to contribute a patch to fix the issue?

None

nyanmisaka, 2024-07-13:
`scale_vaapi` only does scaling, while `zscale` does not only scaling, but also linearization, which is essential for the `tonemap` filter. So mixing `scale_vaapi` and `tonemap` is wrong.

Anuskuss, 2024-07-13:
Can you suggest another filter that does this *linearization*? Because Plex does something similar
```
hwupload,scale_vaapi=w=1920:h=1080:format=p010,hwmap=derive_device=opencl,tonemap_opencl=tonemap=mobius:format=nv12:m=bt709:p=bt709:r=tv,hwmap=derive_device=vaapi:reverse=1,hwupload
```

nyanmisaka, 2024-07-13:
AFAIK currently only `zscale` can do it, that's why `zscale+tonemap` is always used in combination.

Also if your GPU supports `tonemap_vaapi`, you can also try it. But it cannot be fine-tuned.

Anuskuss, 2024-07-15:
`tonemap_vaapi` is even worse than no color conversion at all so in the end I decided to just go the Plex route and use OpenCL instead, even though the colors weren't great either (when compared to `zscale`):
```bash
ffmpeg -i hdr.mp4 -vf format=p010,hwupload,tonemap_opencl=tonemap=mobius:format=nv12:m=bt709:p=bt709:r=tv,hwdownload,format=nv12 thumb-%06d.png
```

> AFAIK currently only `zscale` can do it

Okay so if `scale_vaapi=format=p010,tonemap` is wrong should I turn this issue into a feature request instead (asking for  `linear` support)?

nyanmisaka, 2024-07-15:
This is unlikely to happen in VA-API. It's beyond the capabilities of fixed-function hardware. It's better to implement it in OpenCL or Vulkan via compute kernels.

So as an alternative, you can use the `libplacebo` filter, which requires the Vulkan runtime.

Anuskuss, 2024-07-15:
> you can use the `libplacebo` filter

`libplacebo` sounds interesting but I unfortunately can't use that either.

> This is unlikely to happen in VA-API.

Guess I'll close this then. Thanks for your time.

closed 2024-07-15 by Anuskuss