intel/media-driver #1842: [Bug]:

sebastinas, 2024-08-10:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

In Debian we strip the kernels (i.e., we removed the corresponding file) without source code from source package. This approach worked until now. While trying to update the Debian package to 24.3.1, this approach no longer works.  After removing ` media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.c`,  `media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_l0_xe2_hpg.c`, and `media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_xe2_hpg.c` from the source package, the builds fails with:
```
-- Configuring done (1.5s)                                                                                                                                     
CMake Error at media_driver/media_top_cmake.cmake:505 (add_library):                                                                                           
  Cannot find source file:                                                                                                                                     
                                                                                                                                                               
    /<<PKGBUILDDIR>>/media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.c                                                       
                                                                                                                                                               
  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm                                                                                      
  .ccm .cxxm .c++m .h .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90                                                                                     
  .f95 .f03 .hip .ispc                                                                                                                                         
Call Stack (most recent call first):                                                                                                                           
  media_driver/CMakeLists.txt:95 (include)                                                                                                                     
                                                                                                                                                               
                                                                                                                                                               
CMake Error at media_driver/media_top_cmake.cmake:505 (add_library):                                                                                           
  No SOURCES given to target: iHD_drv_video_SOFTLET_VP
Call Stack (most recent call first):    
  media_driver/CMakeLists.txt:95 (include)

CMake Generate step failed.  Build files cannot be regenerated correctly.
        cd obj-x86_64-linux-gnu && tail -v -n \+0 CMakeCache.txt
```

The build is configured with `-DENABLE_KERNELS=ON` and `-DENABLE_NONFREE_KERNELS=OFF`.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

The Debian package cannot be updated.

### Do you want to contribute a patch to fix the issue?

None

Effieyu0, 2024-08-14:
Hi, @jcescamilla ，
I will check it and fix this build issue.

intel-mediadev, 2024-08-14:
Auto Created VSMGWL-75562 for further analysis.

Effieyu0, 2024-08-15:
@sebastinas 
Which version of the driver are you using for this issue?
I can’t pre-produce the issue on newest open source driver with the following steps:
1.	Deleted xe2_hpg kernel manually
2.	Added -DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF

sebastinas, 2024-08-15:
The issue happens with the 24.3.1 release.

Effieyu0, 2024-08-22:
@sebastinas
We can't re-produce the error in our build, I attached build log and deleted kernel files.

[open_sourec_build_log.txt](https://github.com/user-attachments/files/16702767/open_sourec_build_log.txt)

Delete files as following:
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.c
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.h
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/media_srcs.cmake
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_l0_xe2_hpg.c
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_l0_xe2_hpg.h
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_xe2_hpg.c
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_xe2_hpg.h
        deleted:    media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/media_srcs.cmake

Effieyu0, 2024-08-22:
-- Configuring done (1.5s)                                                                                                                                     
CMake Error at media_driver/media_top_cmake.cmake:505 (add_library):                                                                                           
  Cannot find source file:                          

It seems the error happened during cmake checking:;
Just want to confirm that : have you deleted the make files(media_srcs.cmake) under the following paths:
media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/
media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch

sebastinas, 2024-08-23:
Steps to reproduce:
```
wget https://github.com/intel/media-driver/archive/refs/tags/intel-media-24.3.1.tar.gz
tar xaf intel-media-24.3.1.tar.gz
media-driver-intel-media-24.3.1
rm media_driver/agnostic/Xe_M/Xe_HPM/codec/kernel/XE_HPM_VC1_OLP.c
rm media_driver/agnostic/Xe_M/Xe_HPM/codec/kernel/Xe_Hpm_Film_Grain.c
rm media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_Film_Grain.c
rm media_driver/agnostic/Xe_M/Xe_XPM/codec/kernelisa/Xe_XPM_VC1_OLP.c
rm media_driver/agnostic/Xe_M/Xe_XPM/vp/kernel/cmfcpatch/igvpkrn_xe_xpm_cmfcpatch.c
rm media_driver/agnostic/Xe_M/Xe_XPM/vp/kernel/igvpkrn_isa_xe_xpm.c
rm media_driver/agnostic/Xe_M/Xe_XPM/vp/kernel/igvpkrn_xe_xpm.c
rm media_driver/agnostic/Xe_M/Xe_XPM_plus/codec/kernel/Xe_XPM_plus_Film_Grain.c
rm media_driver/agnostic/Xe_M/Xe_XPM_plus/vp/kernel/cmfcpatch/igvpkrn_xe_xpm_plus_cmfcpatch.c
rm media_driver/agnostic/Xe_M/Xe_XPM_plus/vp/kernel/igvpkrn_xe_xpm_plus.c
rm media_driver/agnostic/gen11/codec/kernel/igcodeckrn_g11.c
rm media_driver/agnostic/gen11_icllp/codec/kernel/igcodeckrn_g11_icllp.c
rm media_driver/agnostic/gen11_icllp/vp/kernel/igvpkrn_g11_icllp.c
rm media_driver/agnostic/gen11_icllp/vp/kernel/igvpkrn_isa_g11_icllp.c
rm media_driver/agnostic/gen12/codec/kernel/igcodeckrn_g12.c
rm media_driver/agnostic/gen12/codec/kernelisa/*.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/cmfc/igvpkrn_g12_tgllp_cmfc.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/cmfccmlpch/igvpkrn_g12_tgllp_cmfccmlpch.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/cmfcpatch/igvpkrn_g12_tgllp_cmfcpatch.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/igvpkrn_g12_tgllp.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/igvpkrn_isa_g12_tgllp.c
rm media_driver/agnostic/gen12_tgllp/vp/kernel/swsb/igvpkrn_g12_tgllp_swsb.c
rm media_driver/agnostic/gen8/codec/kernel/igcodeckrn_g8.c
rm media_driver/agnostic/gen8/vp/kernel/igvpkrn_g8.c
rm media_driver/agnostic/gen9/codec/kernel/igcodeckrn_g9.c
rm media_driver/agnostic/gen9/vp/kernel/igvpkrn_g9.c
rm media_driver/agnostic/gen9/vp/kernel/igvpkrn_isa_g9.c
rm media_driver/agnostic/gen9_bxt/codec/kernel/igcodeckrn_g9_bxt.c
rm media_driver/agnostic/gen9_cml/vp/kernel/igvpkrn_g9_cml.c
rm media_driver/agnostic/gen9_cml/vp/kernel/tgp/igvpkrn_g9_cml_tgp.c
rm media_driver/agnostic/gen9_cmpv/vp/kernel/igvpkrn_g9_cmpv.c
rm media_driver/agnostic/gen9_kbl/codec/kernel/igcodeckrn_g9_kbl.c
rm media_softlet/agnostic/Xe_R/Xe_HPG_Base/vp/kernel/cmfcpatch/igvpkrn_xe_hpg_cmfcpatch.c
rm media_softlet/agnostic/Xe_R/Xe_HPG_Base/vp/kernel/igvpkrn_isa_xe_hpg.c
rm media_softlet/agnostic/Xe_R/Xe_HPG_Base/vp/kernel/igvpkrn_xe_hpg.c
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.c
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_l0_xe2_hpg.c
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_xe2_hpg.c
mkdir build
cd build
cmake -DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF ..
```

Effieyu0, 2024-08-26:
@sebastinas  From you message, you just deleted source files:
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/cmfcpatch/igvpkrn_xe2_hpg_cmfcpatch.c
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_l0_xe2_hpg.c
rm media_softlet/agnostic/Xe_R/Xe2_HPG/vp/kernel/igvpkrn_xe2_hpg.c
But didn't delete the make file media_srcs.cmake, so the media_top_cmake.cmake still check files.
You also need to deleted media_srcs.cmake  in the same folder or use the above fix code to  exclude kernel folders 
when -DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF

sebastinas, 2024-08-27:
I see. I consider that a regression from how it is handled with other non-free kernels though. This was not necessary before.

Effieyu0, 2024-08-29:
@sebastinas
Could you help to verify the issue is fixed by the commit https://github.com/intel/media-driver/commit/d9e615e00bbd6c69eb8d766dd13b9d31e0802248.
If so,  I will close the issue.
Thanks.

sebastinas, 2024-08-29:
This commit works for me. Thanks.

Effieyu0, 2024-08-30:
> This commit works for me. Thanks.
Thanks for verifying. I will close this bug.

closed 2024-08-30 by Effieyu0