intel/media-driver #1843: [Bug]: vaQueryVideoProcPipelineCaps: filter_flags always 0  in struct VAProcPipelineCaps

z0Kng, 2024-08-14:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Shouldn't it be possible, as described in the [documentation](https://intel.github.io/libva/structVAProcPipelineParameterBuffer.html#abb95e119ed7f841f71b2afbec2104784), to query the supported filter_flags from the driver with vaQueryVideoProcPipelineCaps?

However, filter_flags is always 0:
https://github.com/intel/media-driver/blob/5641059bea7098ab0abd40dc9f3d13f1c7797013/media_driver/linux/common/ddi/media_libva.cpp#L6675

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Its needed for the gstreamer va plugin implementation of interpolation methods, see: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/7301

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-08-16:
Auto Created VSMGWL-75618 for further analysis.

z0Kng, 2024-08-29:
Hi @VincentCheungKokomo thanks for the detailed answer. I had already hardcoded it, I just wanted to check if the hardware supports the different interpolation methods. So is it safe to assume that all hardware supports all interpolation and scaling methods?

VincentCheungKokomo, 2024-08-29:
> Hi @VincentCheungKokomo thanks for the detailed answer. I had already hardcoded it, I just wanted to check if the hardware supports the different interpolation methods. So is it safe to assume that all hardware supports all interpolation and scaling methods?

Yes, hardware supports all interpolation and scaling methods.

closed 2024-08-29 by z0Kng