intel/media-driver #1844: [Feature]: Braswell support (Celeron N3050)

Quackdoc, 2024-08-17:
### What Feature?

Told to to use iHD for braswell for chrome support https://issues.chromium.org/issues/40279587#comment22

```
The i965 driver is no longer actively maintained [1], preventing new PR submissions to its repository.

I strongly recommend using the iHD driver instead [2]. The Braswell iGPU shares the same architecture with Broadwell, please try it with iHD driver. If not currently supported, please file an issue with the iHD project.

Please check the iHD supported platforms:
```

### What's the usage scenario would be benifited?

Playback, Web browser

### Do you want to contribute a patch to develop this feature?

No.

nyanmisaka, 2024-08-17:
According to hw info in intel-vaapi-driver, [Braswell/CherryView](https://github.com/intel/intel-vaapi-driver/blob/06c9a893581c975c365c2281a1e597197b558085/src/i965_pciids.h#L132-L135) is a mixture of Gen8(Broadwell) and Gen9(Skylake). Maybe you can try adding the DIDs to [Gen8 DDI](https://github.com/intel/media-driver/blob/1a3ae6226a6f80e4bcc056f56706d04d0acf7a37/media_driver/linux/gen8/ddi/media_sysinfo_g8.cpp#L243) and see. But anyway the [README](https://github.com/intel/media-driver/blob/master/README.md#supported-platforms) says that Braswell is not a platform supported by iHD.

- https://github.com/intel/intel-vaapi-driver/blob/06c9a893581c975c365c2281a1e597197b558085/src/i965_device_info.c#L284-L330
- https://github.com/intel/intel-vaapi-driver/blob/06c9a893581c975c365c2281a1e597197b558085/src/i965_device_info.c#L239-L281

Quackdoc, 2024-08-17:
I did try patching the file to add support for the igpu, it seems like it almost works, mpv will decode a frame or maybe two, but then the gpu hangs with the below eeror

```
resetting vcs0 preemtpion time out
mpv[PID] context reset due to gpu hang
GPU HANG: ecode 8:4:a01fffff, in mpv [PID]
```

it would indeed be nice for iHD to support braswell since it seems like chrome will no longer work with it, and hasvk + zink-video seems like it won't be viable any time soon

colaas, 2024-12-23:
Is Braswell support ever going to be added?

RJVB, 2025-08-14:
>  But anyway the [README](https://github.com/intel/media-driver/blob/master/README.md#supported-platforms) says that Braswell is not a platform supported by iHD.

Actually it doesn't, not explicitly. If Braswell is based on some kind of mix of Broadwell and Skylake (or "based on Broadwell" according to https://en.wikipedia.org/wiki/Intel_Graphics_Technology) one would assume that it should/might be supported....

(Fortunately Firefox has no troubles supporting the older `intel-vaapi-driver` ;) )

XinfengZhang, 2025-08-15:
Braswell has similar ip with Cherry Trail, is gen8 low power sku. iHD start from gen8, but only support BDW as the first platform.

RJVB, 2025-08-15:
On Friday August 15 2025 01:41:56 Zhang, Xinfeng wrote:
>XinfengZhang left a comment (intel/media-driver#1844)
>
>Braswell has similar ip with Cherry Trail, is gen8 low power sku. iHD start from gen8, but only support BDW as the first platform. 

The thing is that you read everywhere that Braswell is based on Broadwell, which, if true, means it can't be older. I suppose it can be an unsupported lower-power version - I lack the background to see why a lower power consumption would be a problem from a driver-support point of view.

But I stand by my opinion that the list as it is presented doesn't exclude the architecture. It would have to be stipulated that Braswell isn't supported, or that anything not on the list (including derivative (?) architectures) are not supported.

Or just remove any overlap between this and the older driver (and add a note about that) ;)

I guess it's not like us Braswell users stand much to gain even if it were trivial to add support (and possibly even lose, judging from the complexity of the code)?