intel/media-driver #1852: [Bug]: AV1 decoding fails with free kernel, works with non-free

sebastianas, 2024-09-10:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

> $ ffmpeg -y -hwaccel vaapi -i sample.av1.mp4  -f null -
> ffmpeg version 7.0.2-3 Copyright (c) 2000-2024 the FFmpeg developers
>   built with gcc 14 (Debian 14.2.0-2)
>…
>   libavutil      59.  8.100 / 59.  8.100
>   libavcodec     61.  3.100 / 61.  3.100
>   libavformat    61.  1.100 / 61.  1.100
>   libavdevice    61.  1.100 / 61.  1.100
>   libavfilter    10.  1.100 / 10.  1.100
>   libswscale      8.  1.100 /  8.  1.100
>   libswresample   5.  1.100 /  5.  1.100
>   libpostproc    58.  1.100 / 58.  1.100
> Input #0, matroska,webm, from 'sample.av1.mkv':
>   Metadata:
>     ENCODER         : Lavf61.1.100
>   Duration: 00:00:01.00, start: 0.000000, bitrate: 11 kb/s
>   Stream #0:0: Video: av1 (libdav1d) (Main), yuv420p(tv, progressive), 1920x1080, SAR 1:1 DAR 16:9, 25 fps, 25 tbr, 1k tbn
>       Metadata:
>         ENCODER         : Lavc61.3.100 libsvtav1
>         DURATION        : 00:00:01.000000000
> Stream mapping:
>   Stream #0:0 -> #0:0 (av1 (native) -> wrapped_avframe (native))
> Press [q] to stop, [?] for help
> [av1 @ 0x562e9b783240] Failed to end picture decode issue: 23 (internal decoding error).
> [av1 @ 0x562e9b783240] HW accel end frame fail.
> [vist#0:0/av1 @ 0x562e9b7814c0] [dec:av1 @ 0x562e9b782b40] Error submitting packet to decoder: Input/output error
> Segmentation fault

It works without vaapi.
It woirks with vaapi and non-free kernels.
The media files is provided in the debug section.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

I was never able to play a AV1 encoded media on this hardware as long as I have it. I initially worked around it by disabling hardware acceleration hoping it will resolve on its own but didn't so far. The free kernels are shipped in Debian by default. I just learned today (while debugging) that non-free kernels in a different package also exists.

### Debug Information

intel-media-va-driver:amd64 24.2.5+dfsg1-1

$ ls /dev/dri/
by-path  card0  renderD128

https://github.com/user-attachments/assets/3731fef0-096b-4750-9681-afea1caeea52

[libva_trace.log.txt](https://github.com/user-attachments/files/16942892/libva_trace.log.txt)
[vainfo.txt](https://github.com/user-attachments/files/16942893/vainfo.txt)

00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

Jexu, 2024-09-11:
Please check the details in https://github.com/intel/media-driver/issues/1832

This is caused by lack of filmgram kernel with free kernel build. AV1 will not work without it on tgl, but there will be no kernel dependency anymore from mtl+.

sebastianas, 2024-09-17:
Didn't realize this is a duplicate of the other bug.
What I don't understand after reading the other report, is this a situation that will change soon or unlikely to change at all?  If it won't change, wouldn't it be sane to remove the acceleration from the "free" driver (and keep it only in the "non-free" driver) rather than keeping the support enabled keep crashing?

Jexu, 2024-09-18:
Will not change at all on tgl/dg1/dg2, yeah we need to remove it from 'free' driver feature list for these platforms.

XinfengZhang, 2024-09-18:
@Jexu , how about just remove film grain feature , still support AV1d?

Jexu, 2024-09-18:
It needs a lot of works to refine the code to separate filmgrain and I think it will not be priorized since driver has moved to fix function from mtl.
Btw, filmgrain apply is frame leve syntax, it means some of the frame could apply filmgrain, some of them doesn;t need. You cannot know if frame sequences need it before decoding.

PR #1853 (2024-10-11): [Decode] Remove av1 decode from feature list on some platforms
Fixes: #1852, #1832
Av1 decode needs filmgrain kernel support, and free kernel build will not support av1 decode.
commit 725f3bf5 (2024-10-11): [Decode] Remove av1 decode from feature list on some platforms

Fixes: #1852, #1832
Av1 decode needs filmgrain kernel support, and free kernel build will
not support av1 decode.
closed 2024-10-11 by intel-mediadev