intel/media-driver #1857: [Bug]: ffmpeg-vaapi decode 10bit vp9 have lots case fail on LNL by md5 mismatch

Hangjie22Coder, 2024-09-30:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

VERSION="20.04.5 LTS
kernel:5.15.47+prerelease3723

libva:https://github.com/intel/libva/commit/0d971e918da2af176f158cad3948b1c88db3d55b
libva-utils:https://github.com/intel/libva-utils/commit/43555d6b3b1a90f6519fdab6a463d1d56398831a
gmmlib:https://github.com/intel/gmmlib/commit/c2dbdedeb71de19c70435d507f71b83540afbf13

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

cmd line:
gst-launch-1.0 -vf filesrc location=input.ivf ! ivfparse ! vp9parse ! vaapivp9dec ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=P010_10LE ! checksumsink2 qos=false file-checksum=false frame-checksum=false plane-checksum=false eos-after=8 dump-output=true dump-location=output.yuv

result:
md5:expect = 61725148e9b72f5c5204831889e5e19e
md5:actual = d8a56e7a32aefb8bd81716807b291583

If iHD revert to https://github.com/intel/media-driver/commit/b3489939f083a56a373fa66a368ea4c1bd11ed04, the issue will disappear.

### Do you want to contribute a patch to fix the issue?

None

Jexu, 2024-10-09:
It is regression or not? Could you help get the first bad commit id?

Hangjie22Coder, 2024-10-11:
Thank you for your comment. This bug has been resolved internally on our side, so there is no need to investigate further. We won't be pursuing the first bad commit as the issue no longer persists.

closed 2024-10-11 by Hangjie22Coder