intel/media-driver #1860: [Bug]: Lack of hardware decoding capability on the Core i7 8 Gen.

fxzxmicah, 2024-10-12:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

Lack of hardware decoding capabilities, such as AV1.
Everything works fine with the old intel-vaapi-driver.
CPU: i7 8750H
System: fedora 40
Browser: Firefox

### What's the usage scenario when you are seeing the problem?

Web browser

### Do you want to contribute a patch to fix the issue?

None

fxzxmicah, 2024-10-12:
I saw the table of supported decoding formats. So is this missing decoder temporary or intentional?

Jexu, 2024-10-14:
Do you mean vainfo lists nothing? How about web decoding if still working?

fxzxmicah, 2024-10-14:
> Do you mean vainfo lists nothing? How about web decoding if still working?

Your readme table states that Core i7 8 Gen does not support AV1 decoding, but in fact it does.
When using the old driver, vainfo lists more decoding formats than the new driver.

nyanmisaka, 2024-10-14:
The latest GPUs released after 2020 usually support AV1 hardware decoding, which are 11th Gen Intel Tiger Lake (TGLx), NVIDIA RTX 3000 series and AMD RDNA2 (RX 6000). So it is obvious that 8th Gen Intel iGPU does not support AV1 decoding.

You can also find out from the official GPU programming manual that Coffee Lake (KBLx) does not support AV1. https://www.intel.com/content/www/us/en/docs/graphics-for-linux/developer-reference/1-0/overview.html

fxzxmicah, 2024-10-14:
OLD:
```
Trying display: wayland
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/iHD_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/iHD_drv_video.so
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
libva info: Trying to open /usr/lib64/dri-nonfree/i965_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/i965_drv_video.so
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.4.0.pre1 (2.4.0.pre1)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSlice
      VAProfileVP9Profile2            :	VAEntrypointVLD
```
NEW:
```
Trying display: wayland
libva info: VA-API version 1.21.0
libva info: Trying to open /usr/lib64/dri-nonfree/iHD_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/iHD_drv_video.so
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_21
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.21 (libva 2.21.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.2.5 ()
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

fxzxmicah, 2024-10-14:
> The latest GPUs released after 2020 usually support AV1 hardware decoding, which are 11th Gen Intel Tiger Lake (TGLx), NVIDIA RTX 3000 series and AMD RDNA2 (RX 6000). So it is obvious that 8th Gen [...]

That might be something I got wrong. Sorry for that.

nyanmisaka, 2024-10-14:
You may have mistaken VC1 for AV1. But you are using the so-called free version of the iHD driver (media kernel/shaders are stripped by some linux distro), which means it only has very limit features.

https://github.com/intel/media-driver#components-and-features

fxzxmicah, 2024-10-14:
Well, that should be the problem. Sorry to bother you.

closed 2024-10-14 by fxzxmicah