intel/media-driver #1863: [Bug]: Regression, BGRx no longer supported in vapostproc

z0Kng, 2024-10-17:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

After update to intel-media  24.3.3-i992.el9_4 from 24.2.5-i947.el9_4  the gstreamer va plugins vapostproc, no longer supports BGRx. I tracked it back to this commit: https://github.com/intel/media-driver/commit/7c9645864d1437480a347d7fdac502ffe21d5716

The gpu's used are flex 170 and flex 140.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

We are building a vdi solution, the framebuffer of the intel gpu is in BGRx format and we need to video encode it. It works in the old media driver.

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-10-18:
Auto Created VSMGWL-76592 for further analysis.

z0Kng, 2024-10-28:
Seems to be fixed in a newer commit.

closed 2024-10-28 by z0Kng