intel/media-driver #1867: [Bug]: VP format conversion to VUYA results in pink artifacts.

z0Kng, 2024-10-29:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

None

### What happened?

Converting for e.g. BGRx to VUYA using the gstreamer va plugins vapostproc, results in pink artifacts while converting to NV12 is fine. It only appears if the image height is not a multiple of 16.
Example gstreamer pipeline:
``` 
gst-launch-1.0 videotestsrc pattern=ball flip=true ! video/x-raw,format=BGRx,width=1920,height=1080 ! vapostproc ! "video/x-raw(memory:VAMemory),format=VUYA" ! vah265lpenc ! h265parse ! matroskamux ! filesink location=test.mkv
```
[image]

The same pipeline with height=1072 works fine.

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

We are currently developing an opensource vdi solution using Flex vGPUs for graphics acceleration and video stream encoding. For non chroma subsampling encoding, we need to convert BGRx into VUYA.

### Debug Information

1. vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.4.2 (18b64f08b)
2. exists
3. Intel Corporation ATS-M [Data Center GPU Flex 170] [8086:56c0] (rev 08)
4. [vainfo.log](https://github.com/user-attachments/files/17554685/vainfo.log)
5. [libva_trace.log](https://github.com/user-attachments/files/17554748/libva_trace.log)

### Do you want to contribute a patch to fix the issue?

None

z0Kng, 2024-11-20:
I could also verify the issue on a Flex 140 with:

```
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.4.3 (bd984fbb1)
```

and Intel® Core™ Ultra 7 155H with:

```
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.3.4 ()
```

intel-mediadev, 2024-11-28:
Auto Created VSMGWL-77288 for further analysis.

jiafengy1, 2024-12-12:
[image]
unable to reproduce, could you share the resource that needed to reproduce this issue?

z0Kng, 2024-12-12:
As far as I'm aware, it depends on the hardware whether the gstreamer va plugin has `vah265enc` or `vah265lpenc`. From what I noticed, older intel gpus tend to have `vah265enc` and newer `vah265lpenc`. I think it's either from ICL or MTL onwards. 
However, this issue only happens on newer generations, since I was unable to reproduce it on an 8th Gen Intel CPU.

jiafengy1, 2024-12-18:
Seems that both VP input and output are fine.

z0Kng, 2025-02-04:
Any update on this? We could meanwhile reproduce the issue on even more hardware and different distros.