intel/media-driver #1871: [Bug]: OBS QuickSync encoding issues with drivers 32.0.101.6127/32.0.101.6044  - 32.0.101.6130/32.0.101.6048

maginhart, 2024-10-30:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

I'm using Intel Arc A380 in a separate pc for encoding purposes.
After installing **32.0.101.6127/32.0.101.6044** drivers the obs qsv encoding performance regresses. 
Average time to render frame jumped from usual 2.5-3.0 ms to 10 ms with one local recording session. (QuickSync h264/hevc/av1)
The last three drivers have this issue.
Rolling back to at least **32.0.101.6083/32.0.101.5736** resolves this issue.

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery, Content Creation, Game Streaming

### Do you want to contribute a patch to fix the issue?

None

bai-isaac, 2024-10-31:
hi maginhart, which OS do you use?
This issue is for all codec(AV1/HEVC/AVC)?

maginhart, 2024-10-31:
Windows 11 23H2 yes all codecs.

maginhart, 2024-10-31:
I've launched fresh obs OBS 30.2.3 with only one scene element video capture device.
Bellow you can see the difference between drivers 32.0.101.6083/32.0.101.5736 and 32.0.101.6127/32.0.101.6044

[obs_old_driver_2024-10-31 11-33-43.txt](https://github.com/user-attachments/files/17585758/obs_old_driver_2024-10-31.11-33-43.txt)
[image]

[obs_new_driver_2024-10-31 11-19-24.txt](https://github.com/user-attachments/files/17585759/obs_new_driver_2024-10-31.11-19-24.txt)
[image]

maginhart, 2024-11-13:
I've tried the latest driver 32.0.101.6297 still the same problem.