intel/media-driver #1879: [Bug]: SIGSEGV when seeking in certain videos on 285K (ARL-S)

database64128, 2024-11-15:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

1. Play certain videos in mpv.
2. Seek forward and backward.
3. mpv may crash. See the attached backtrace below.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

Video playback in mpv.

### Debug Information

- OS: Arch Linux
- `intel-media-driver`: 24.4.2-1
- `vpl-gpu-rt`: 24.4.2-1

```console
#0  MediaStatusReport::GetCompletedCount (this=0x0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_softlet/agnostic/common/shared/statusreport/media_status_report.h:123                                                                                      
123	        if (m_completedCount == nullptr)
[Current thread is 1 (Thread 0x74b729c006c0 (LWP 220735))]
(gdb) bt
#0  MediaStatusReport::GetCompletedCount (this=0x0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_softlet/agnostic/common/shared/statusreport/media_status_report.h:123
#1  decode::Av1PipelineG12::GetCompletedReport (this=0x6535d8e785e0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_driver/media_softlet/agnostic/gen12/codec/hal/dec/av1/pipeline/decode_av1_pipeline_g12.cpp:254
#2  DecodeAvcPipelineAdapterXe_Lpm_Plus_Base::GetCompletedReport (this=0x6535d8e5e170)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus_base/codec/hal/dec/avc/pipeline/decode_avc_pipeline_adapter_xe_lpm_plus_base.cpp:124
#3  0x000074b719c9007f in DdiDecodeFunctions::StatusReport (this=this@entry=0x74b6e81f33f0, mediaCtx=mediaCtx@entry=0x74b6e8012d30, decoder=0x6535d8e5e170, surface=surface@entry=0x6535d8e3fda0)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_softlet/linux/common/codec/ddi/dec/ddi_decode_functions.cpp:1049
#4  0x000074b719c91bf4 in DdiDecodeFunctions::StatusCheck (this=0x74b6e81f33f0, mediaCtx=0x74b6e8012d30, surface=0x6535d8e3fda0, surfaceId=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.2/media_softlet/linux/common/codec/ddi/dec/ddi_decode_functions.cpp:1031
#5  0x000074b74ccdcced in vaSyncSurface (dpy=dpy@entry=0x74b6e80dad90, render_target=<optimized out>) at ../libva/va/va.c:1668
#6  0x00006535b5688cd9 in mapper_map (mapper=0x74b6e8483020) at ../mpv/video/out/hwdec/hwdec_vaapi.c:295
#7  0x00006535b560979a in ra_hwdec_mapper_map (mapper=0x74b6e8483020, img=0x74b6e9790110) at ../mpv/video/out/gpu/hwdec.c:172
#8  0x00006535b562b7af in hwdec_acquire (gpu=<optimized out>, frame=0x74b729bfb0f8) at ../mpv/video/out/vo_gpu_next.c:567
#9  0x000074b750958f00 in acquire_frame (pass=0x74b729bfaf90, frame=0x74b729bfb0f8, acquired=0x74b729bfbca2) at src/renderer.c:2881
#10 pass_init (pass=pass@entry=0x74b729bfaf90, acquire_image=acquire_image@entry=true) at src/renderer.c:3014
#11 0x000074b750960af1 in pl_render_image (rr=rr@entry=0x74b6e8033a40, pimage=pimage@entry=0x74b6e80c8700, ptarget=ptarget@entry=0x74b729bfef30, params=params@entry=0x74b729bfeb30) at src/renderer.c:3138
#12 0x000074b750962250 in pl_render_image_mix (rr=<optimized out>, images=images@entry=0x74b729bfe9a0, ptarget=ptarget@entry=0x74b729bfef30, params=params@entry=0x74b729bfeb30) at src/renderer.c:3688
#13 0x00006535b562d200 in draw_frame (vo=<optimized out>, frame=<optimized out>) at ../mpv/video/out/vo_gpu_next.c:1124
#14 0x00006535b56238e5 in render_frame (vo=0x6535d8b7b1e0) at ../mpv/video/out/vo.c:973
#15 vo_thread (ptr=<optimized out>) at ../mpv/video/out/vo.c:1105
#16 0x000074b74cb2d39d in start_thread (arg=<optimized out>) at pthread_create.c:447
#17 0x000074b74cbb249c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

### Do you want to contribute a patch to fix the issue?

None

database64128, 2024-11-28:
Update: After upgrading `intel-media-driver` from 24.4.2-1 to 24.4.3-1, VAAPI playback is completely broken. The driver crashes mpv immediately on startup. Here's the new backtrace:

```console
Program terminated with signal SIGSEGV, Segmentation fault.
Downloading 52.82 K source file /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp
#0  vp::VpAllocator::DestroyVpSurface (this=0x76e1bc481b90, surface=@0x76e1ebffd0a8: 0x76e1bc237660, deferredDestroyed=false, flags=...)                        
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
508	        int64_t currentSize = static_cast<int64_t>(surface->osSurface->OsResource.pGmmResInfo ? surface->osSurface->OsResource.pGmmResInfo->GetSizeAllocation() : 0);
[Current thread is 1 (Thread 0x76e1ebfff6c0 (LWP 3737))]
(gdb) bt
#0  vp::VpAllocator::DestroyVpSurface (this=0x76e1bc481b90, surface=@0x76e1ebffd0a8: 0x76e1bc237660, deferredDestroyed=false, flags=...)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
#1  0x000076e1ddf93313 in vp::SwFilterPipe::Clean (this=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/sw_filter_pipe.cpp:409
#2  0x000076e1de350e43 in vp::VpObjAllocator<vp::SwFilterPipe>::Destory (this=0x76e1bc4c96f0, obj=@0x76e1bc4f44e0: 0x76e1bc47c010)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.h:84
#3  vp::SwFilterPipeFactory::Destory (this=0x76e1bc4c96e8, swFilterPipe=@0x76e1bc4f44e0: 0x76e1bc47c010)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.cpp:298
#4  vp::VpPipeline::ExecuteSingleswFilterPipe(vp::VpSinglePipeContext*, vp::SwFilterPipe*&, vp::PacketPipe*, vp::VpFeatureManagerNext*) [clone .constprop.0]
    (this=this@entry=0x76e1bc23c800, singlePipeCtx=<optimized out>, pipe=@0x76e1bc4f44e0: 0x76e1bc47c010, featureManagerNext=featureManagerNext@entry=0x76e1bc4d5fc0, pPacketPipe=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:692
#5  0x000076e1ddfe7e21 in vp::VpPipeline::ExecuteVpPipeline (this=0x76e1bc23c800)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:589
#6  0x000076e1de00f035 in virtual thunk to VpPipelineAdapterXe_Lpm_Plus::Execute(VPHAL_RENDER_PARAMS*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/vp/hal/pipeline/vp_pipeline_adapter_xe_lpm_plus.h:73
#7  0x000076e1ddfe8538 in VpPipelineAdapter::Render (this=0x76e1bc4b1638, pcRenderParams=0x76e1bc47b3b0)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline_adapter.cpp:177
#8  0x000076e1de014d1b in virtual thunk to VpPipelineAdapterXe_Lpm_Plus::Render(VPHAL_RENDER_PARAMS const*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/vp/hal/pipeline/vp_pipeline_adapter_xe_lpm_plus.h:53
#9  0x000076e1ddc8bc45 in DdiVpFunctions::EndPicture (this=<optimized out>, ctx=0x76e1bc0db1a0, context=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:534
#10 0x000076e1ddc90201 in DdiVpFunctions::ProcessPipeline
    (this=0x76e1bc1f3700, vaDrvCtx=0x76e1bc0db1a0, ctxID=3489660928, srcSurface=2, srcRect=0x76e1ebffdb18, dstSurface=1, dstRect=0x76e1ebffdb20)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:4472
#11 0x000076e1ddc49b4a in MediaLibvaInterfaceNext::PutImage
    (ctx=0x76e1bc0db1a0, surface=1, image=<optimized out>, srcX=0, srcY=0, srcWidth=2, srcHeight=2, destX=0, destY=0, destWidth=2, destHeight=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/ddi/media_libva_interface_next.cpp:1867
#12 MediaLibvaInterface::PutImage
    (ctx=0x76e1bc0db1a0, surface=1, image=<optimized out>, src_x=0, src_y=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_driver/linux/common/ddi/media_libva_interface.cpp:384
#13 0x000076e2129eb83f in vaPutImage
    (dpy=0x76e1bc0db030, surface=surface@entry=1, image=0, src_x=src_x@entry=0, src_y=src_y@entry=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2) at ../libva/va/va.c:1893
#14 0x000076e21363c37d in vaapi_unmap_frame (hwfc=0x76e1bc4a9ec0, hwmap=<optimized out>) at libavutil/hwcontext_vaapi.c:782
#15 0x000076e213631dc3 in ff_hwframe_unmap (opaque=<optimized out>, data=0x76e1bc47c900 "\300\233\037\274\341v") at libavutil/hwcontext.c:717
#16 0x000076e21361c9b7 in buffer_replace (dst=<optimized out>, src=0x0) at libavutil/buffer.c:133
#17 av_buffer_unref (buf=<optimized out>) at libavutil/buffer.c:144
[... cut ...]
#35 0x00006023cdb03941 in control (vo=0x6023e9bb5ad0, request=<optimized out>, data=0x7ffd5ed9a440) at ../mpv/video/out/vo_gpu_next.c:1558
#36 0x00006023cdae0dde in run_control (p=0x7ffd5ed9a390) at ../mpv/video/out/vo.c:652
#37 0x00006023cda5bcc2 in mp_dispatch_queue_process (queue=0x6023e9bec510, timeout=<optimized out>) at ../mpv/misc/dispatch.c:300
#38 0x00006023cdaea20e in vo_thread (ptr=<optimized out>) at ../mpv/video/out/vo.c:1100
#39 0x000076e21283d39d in start_thread (arg=<optimized out>) at pthread_create.c:447
#40 0x000076e2128c249c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

q234rty, 2025-01-08:
For the record, I can repro the original issue with a similar backtrace, albeit with hevc and `DecodeHevcPipelineAdapterXe_Lpm_Plus_Base`

ye-discernis, 2026-09-24:
Same crash here on Arrow Lake-HX (Legion 5, Ubuntu 26.04). I think I've caught the cause of the NULL `m_statusReport` that @XinfengZhang asked about: **the decode context is being destroyed on another thread while `vaSyncSurface` runs on one of its surfaces.**

**Environment**
- CPU/iGPU: Intel Core Ultra (Arrow Lake-HX), `Intel(R) Graphics (ARL)`; the NVIDIA dGPU was not involved
- intel-media-va-driver-non-free 26.1.2+ds1-1, libva 2.23.0, FFmpeg/libavcodec 8.0.1, libplacebo 7.360, mpv 0.41.0
- Kernel 7.0.0-31 (i915), GNOME Wayland
- mpv: `vo=gpu-next`, `gpu-api=vulkan`, `hwdec=vaapi`
- Content: AV1 Main 1920x1080 yuv420p (YouTube webm)

**Kernel log**
```
vo[2020665]: segfault at 10 ip 00007a3fa22fe20c ... error 4 in iHD_drv_video.so[afe20c,7a3fa19b7000+ad4000]
```

**Faulting thread (mpv render thread):** matches the #1879 path, `vaSyncSurface` → driver status check
```
#0  iHD_drv_video.so +0xafe20c     mov 0x50(%rdi),%rax ; mov 0x60(%rax),%rax ; mov 0x10(%rax),%rdx  <- rax == 0
#1  iHD_drv_video.so
#2  iHD_drv_video.so
#3  vaSyncSurface () libva.so.2
#4-6 mpv (vaapi hwdec mapper)
#7  libplacebo  → pl_render_image → pl_render_image_mix
```

**At the same instant, mpv's main thread is inside `vaDestroyContext` for the same decoder:**
```
#0-14 iHD_drv_video.so   (deep in context/pipeline teardown)
#15 vaDestroyContext () libva.so.2
#16-18 libavcodec.so.62  (vaapi hwaccel uninit)
#19 avcodec_free_context ()
#20+ mpv (decoder uninit)
```

The Ubuntu package has no debug symbols (multiverse, not on debuginfod), so the iHD frames are unnamed. The fault pattern (two pointer hops, then a load at +0x10 from NULL, reached from `vaSyncSurface`) is consistent with `DdiDecodeFunctions::StatusReport → GetCompletedReport → m_statusReport->GetCompletedCount()` with `m_statusReport == nullptr`, i.e. the pipeline's status report already torn down by the concurrent `vaDestroyContext`.

This would explain why the reports here cluster around **seeking** and **quitting / switching files**: those are the moments when FFmpeg frees the decoder while the VO thread may still hold and sync a surface from it. The surfaces belong to the hw frames pool and outlive the decode context, so `vaSyncSurface` on them after or during `vaDestroyContext` should be safe. As I read it, the driver's surface→decoder back-reference isn't cleared or locked during context destruction.

**Workaround confirmed:** `hwdec=vaapi-copy`. The decoder thread copies each frame out, so the VO thread never calls `vaSyncSurface` and the race can't happen.

Possible fix direction: in `vaDestroyContext`, clear the decode context pointer on surfaces that reference it (under the media context mutex), and have `DdiDecodeFunctions::StatusCheck` / `StatusReport` treat a NULL or destroyed decoder / `m_statusReport` as "already complete" instead of dereferencing it.