intel/media-driver #1881: [Bug]: FFmpeg/GStreamer linux sharpness_vaapi filter has no effect on TGL/ADL/RPL platforms with 8K input

Hangjie22Coder, 2024-11-20:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

The sharpness_vaapi filter does not apply any visible changes to the output on TGL (Tiger Lake), ADL (Alder Lake), and RPL (Raptor Lake) platforms under Linux when processing 8K resolution input. The MD5 checksum of the input and output files remains identical, indicating that the sharpening process did not occur. This issue affects both FFmpeg and GStreamer.

cmd:
```
ffmpeg -v verbose -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 \
-hwaccel_output_format nv12 -hwaccel_flags allow_profile_mismatch -i input_8k.h265 \
-lavfi 'null' -c:v rawvideo -pix_fmt nv12 -fps_mode passthrough -noautoscale -vframes 10 -y test_vpp_sharepen.yuv
```
```
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 \
-hwaccel_output_format vaapi -v verbose -f rawvideo -pix_fmt nv12 -s:v 7680x4320 \
-i test_vpp_sharepen.yuv -vf 'format=nv12|vaapi,hwupload,sharpness_vaapi=sharpness=63,hwdownload,format=nv12' \
-pix_fmt nv12 -f rawvideo -fps_mode passthrough -an -vframes 10 -y output_vpp.yuv
```
Expected Behavior
The MD5 checksum of `output_vpp.yuv` should differ from `test_vpp_sharepen.yuv`, indicating that the `sharpness_vaapi` filter has modified the video data.

Actual Behavior
The MD5 checksum of both files is identical:
```
md5sum *.yuv:
77e5c303ac03621e1762266edc5e0bfd  test_vpp_sharepen.yuv
77e5c303ac03621e1762266edc5e0bfd  output_vpp.yuv
```

### What's the usage scenario when you are seeing the problem?

Video Analytics

### What impacted?

**Functionality impacted**: `The filter does not apply sharpening to 8K resolution videos, resulting in no visual difference between the input and output.sharpness_vaapi`
**Platforms impacted**: `TGL (Tiger Lake), ADL (Alder Lake), and RPL (Raptor Lake) on Linux.`

### Debug Information

My environment:
kernel: 6.5.0-44-generic
OS: Ubuntu 22.04.4 LTS
```
        ["libva"]="25381db75af9ff7bb15eb08b4bda95d09cf669b0 (2024-09-25 08:25:55 +0000)"
        ["gmmlib"]="c743f06b430fab6debcb9e0bf82131636e4ede7b (2024-10-18 20:42:51 +0530)"
        ["libva-utils"]="908e2014546bf133cdc8a72cc0d3f2518d04cffc (2024-07-16 13:30:52 +0800)"
        ["media-driver"]="bd984fbb1b6e4bb42d7567dd3dd0e5d27406e93d (2024-11-11 09:36:26 +0800)"
        ["vpl_runtime"]="1501585fea24c601d363333afa8d714572538bfe (2024-11-07 00:39:06 +0000)"
        ["vpl_dispatcher"]="5f6bd8a1e753c8f63a3fd8b36894d6968b808a6d (2024-08-22 15:59:59 +0800)"
        ["cartwheel-ffmpeg"]="1cc91258610873f2b71f6e65271fc724c06df3f0 (2024-11-10 10:01:21 +0000)"
        ["openh264"]="8c7008aeb6335e7d36ab0d9a023a63f82a8eaac0 (2024-11-06 14:05:33 +0800)"
        ["cartwheel-gstreamer"]="7d8096d581282b8defa57a5cda580e80dd327b6a (2024-11-10 10:01:56 +0000)"
        ["gst-checksumsink"]="e1994e1b36dea72b3016624ca724ddf6dd6dc986 (2022-10-11 11:26:19 -0400)"
```

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-11-28:
Auto Created VSMGWL-77287 for further analysis.

VincentCheungKokomo, 2025-01-10:
Root cause: For 8k resolution input on Gen12 platform, VEBOX/SFC will be disable,  so it goes to render path. But for gen12 'm_bFallbackIefPatch' is set to true to fallback IEF patch from AVS to SFC, since SFC is disabled for 8k resolution, so sharpness are not performed.
[image]
[image]

Hangjie22Coder, 2025-01-13:
Thank you for the explanation. Based on your response, I understand that sharpness is not supported for 8K resolution on Gen12 platforms due to the disabling of VEBOX/SFC. Given this limitation, I will remove the 8K resolution sharpen test for these three platforms.

Hangjie22Coder, 2025-01-13:
I will consider the issue resolved and will close the bug.

closed 2025-01-13 by Hangjie22Coder