intel/media-driver #1883: [Bug]: GST VA Video output of Color conversion - UYVY -> RGB32 (Default & Low Power) is corrupted

sengguan, 2024-11-26:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Corrupted output file.
-------------- Reproducible Steps: --------------
1.a) LIN_MEDIA-FGST_VA_CC-M-013: Color conversion - UYVY -> RGB32 #gst-launch-1.0 filesrc location=/media/RAWS/UYVY_Prog_Packed_352x288.yuv ! videoparse format=uyvy width=352 height=288 ! vapostproc ! video/x-raw,format=BGRA ! filesink location=/tmp/GSTVA_CC-UYVY_RGB32-014_output.yuv
  b) LIN_MEDIA-FGST_VA_CC-M-LP-012: Color conversion (LowPower) - UYVY -> RGB32 #gst-launch-1.0 filesrc location=/media/RAWS/UYVY_Prog_Packed_352x288.yuv ! videoparse format=uyvy width=352 height=288 ! vapostproc scale-method=256 ! video/x-raw,format=BGRA ! filesink location=/tmp/GSTVA_CC-UYVY_RGB32_LP-014_output.yuv

2. Open the yuv file via VQA or multiviewer

### What's the usage scenario when you are seeing the problem?

Content Creation

### What impacted?

BTL and TWL release get impacted.

### Debug Information

Caused by https://github.com/intel/media-driver/commit/019772d5fc61f36f2e154477037bb6e86fb48f3e
mainly contribute by 
gmmParams.Flags.Info.Linear = 1; in function DdiMedia_CreateImage()

### Do you want to contribute a patch to fix the issue?

None

intel-mediadev, 2024-11-28:
Auto Created VSMGWL-77286 for further analysis.

sengguan, 2025-01-07:
quick fix for current issue
https://github.com/intel-innersource/drivers.gpu.unified/pull/193001

closed 2025-01-07 by sengguan