intel/media-driver #1885: [Bug]:

PaddyMac, 2024-11-27:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

None

### What happened?

I am running Gentoo Linux with Plasma 6.2.3 on Wayland, and I have an Intel Arc A380 graphics card. Mesa 24.3.0, intel-vaapi-driver 2.4.1, Intel media driver 24.4.3, libva 2.22.0, vpl-gpu-rt 24.4.2, libvpl 2.13.0, gmmlib 22.5.4, libva-utils 2.22.0.

Whether I'm running OBS Studio 30.2.3 or the current 31.0.0-rc1 (I haven't tested other versions since I ran into this issue), OBS Studio will crash with a SIGBUS error within a very few minutes of beginning to record or stream a Linux native game. I've only tested a few games which have both Linux native and Windows versions available. I can run the Windows version of a game with Proton in Steam and either record or stream the game with very rare crashes but then switch to the Linux version of the same game and experience repeated crashes often within 1 or 2 minutes of beginning to stream or record.

I used coredumpctl debug to get some basic debug output which I will attach to the report. I filed a report yesterday on gmmlib (issue 125) for this issue because the backtrace initially seemed to indicate gmmlib was where the fault occurred, but I had OBS compiled with LTO enabled. I recompiled it without LTO, and the crash began happening in media-driver. So I think LTO, while not causing the error, may have been causing the debug info to be different. Compiling without LTO enabled also seemed to allow OBS to record for several minutes longer before the crash occurred.

[coredump-media-driver.txt](https://github.com/user-attachments/files/17928556/coredump-media-driver.txt)

### What's the usage scenario when you are seeing the problem?

Content Creation, Game Streaming

### Debug Information

/dev/dri exists.

03:00.0 VGA compatible controller [0300]: Intel Corporation DG2 [Arc A380] [8086:56a5] (rev 05)
06:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Oland PRO [Radeon R7 240/340 / Radeon 520] [1002:6613] (rev 87)

[vainfo.log](https://github.com/user-attachments/files/17928614/vainfo.log)

I have a trace log, but it is 310 MB in size, so probably too big to attach.

### Do you want to contribute a patch to fix the issue?

None

PaddyMac, 2024-12-12:
I'm just checking in to see if you need any additional information since I haven't heard anything in 2 weeks.

XinfengZhang, 2025-04-11:
sorry for late response, actually, I could not get enough information from current log. 
1. could you recompile the driver manually and keep the symbols ? then we could check the back trace to get some useful information.  
2. could you share the dmesg to check whether there are gpu hang messages?
3. if there are gpu hang, could you help to dump the i915_error_state , and share it?