intel/media-driver #1887: [Bug]: Segfault with VAAPI using mpv on Meteor Lake

q234rty, 2024-12-04:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

1. Use `mpv --no-config --hwdec=vaapi` to play a video with VA-API
2. Observe that mpv crashes.

OS is Arch Linux, with intel-media-driver 24.4.3 and intel-gmmlib 22.5.2.
24.4.2 seems to be fine and my bisection points to https://github.com/intel/media-driver/commit/2c51f70987d52c46fb93bb8831043d0197df7aeb

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

gdb `bt`:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  vp::VpAllocator::DestroyVpSurface (this=0x7e27542f3b90, surface=@0x7e27830c8138: 0x7e275471d520, deferredDestroyed=false, flags=...)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
508	       int64_t currentSize = static_cast<int64_t>(surface->osSurface->OsResource.pGmmResInfo ? surface->osSurface->OsResource.pGmmResInfo->GetSizeAllocation() : 0);
[Current thread is 1 (Thread 0x7e27830ca6c0 (LWP 30826))]
(gdb) bt
#0  vp::VpAllocator::DestroyVpSurface (this=0x7e27542f3b90, surface=@0x7e27830c8138: 0x7e275471d520, deferredDestroyed=false, flags=...)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
#1  0x00007e2771f93313 in vp::SwFilterPipe::Clean (this=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/sw_filter_pipe.cpp:409
#2  0x00007e2772350e43 in vp::VpObjAllocator<vp::SwFilterPipe>::Destory (this=0x7e27547066a0, obj=@0x7e2754493240: 0x7e27546d2080)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.h:84
#3  vp::SwFilterPipeFactory::Destory (this=0x7e2754706698, swFilterPipe=@0x7e2754493240: 0x7e27546d2080)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.cpp:298
#4  vp::VpPipeline::ExecuteSingleswFilterPipe(vp::VpSinglePipeContext*, vp::SwFilterPipe*&, vp::PacketPipe*, vp::VpFeatureManagerNext*) [clone .constprop.0]
    (this=this@entry=0x7e2754493270, singlePipeCtx=<optimized out>, pipe=@0x7e2754493240: 0x7e27546d2080, featureManagerNext=featureManagerNext@entry=0x7e27546ffa00, pPacketPipe=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:692
#5  0x00007e2771fe7e21 in vp::VpPipeline::ExecuteVpPipeline (this=0x7e2754493270)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:589
#6  0x00007e277200f035 in virtual thunk to VpPipelineAdapterXe_Lpm_Plus::Execute(VPHAL_RENDER_PARAMS*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/vp/hal/pipeline/vp_pipeline_adapter_xe_lpm_plus.h:73
#7  0x00007e2771fe8538 in VpPipelineAdapter::Render (this=0x7e2754484968, pcRenderParams=0x7e27546d1400)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline_adapter.cpp:177
#8  0x00007e2772014d1b in virtual thunk to VpPipelineAdapterXe_Lpm_Plus::Render(VPHAL_RENDER_PARAMS const*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/agnostic/Xe_M_plus/Xe_LPM_plus/vp/hal/pipeline/vp_pipeline_adapter_xe_lpm_plus.h:53
#9  0x00007e2771c8bc45 in DdiVpFunctions::EndPicture (this=<optimized out>, ctx=0x7e275432b8d0, context=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:534
#10 0x00007e2771c90201 in DdiVpFunctions::ProcessPipeline
    (this=0x7e27544499f0, vaDrvCtx=0x7e275432b8d0, ctxID=3489660928, srcSurface=2, srcRect=0x7e27830c8ba8, dstSurface=1, dstRect=0x7e27830c8bb0)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:4472
#11 0x00007e2771c49b4a in MediaLibvaInterfaceNext::PutImage
    (ctx=0x7e275432b8d0, surface=1, image=<optimized out>, srcX=0, srcY=0, srcWidth=2, srcHeight=2, destX=0, destY=0, destWidth=2, destHeight=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_softlet/linux/common/ddi/media_libva_interface_next.cpp:1867
#12 MediaLibvaInterface::PutImage
    (ctx=0x7e275432b8d0, surface=1, image=<optimized out>, src_x=0, src_y=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.3/media_driver/linux/common/ddi/media_libva_interface.cpp:384
#13 0x00007e27a9abd83f in vaPutImage
    (dpy=0x7e275432b760, surface=surface@entry=1, image=0, src_x=src_x@entry=0, src_y=src_y@entry=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2)
    at ../libva/va/va.c:1893
#14 0x00007e27aa63c37d in vaapi_unmap_frame (hwfc=0x7e27546e0c80, hwmap=<optimized out>) at libavutil/hwcontext_vaapi.c:782
#15 0x00007e27aa631dc3 in ff_hwframe_unmap (opaque=<optimized out>, data=0x7e2754450c00 "@PET'~") at libavutil/hwcontext.c:717
#16 0x00007e27aa61c9b7 in buffer_replace (dst=<optimized out>, src=0x0) at libavutil/buffer.c:133
#17 av_buffer_unref (buf=<optimized out>) at libavutil/buffer.c:144
#18 av_buffer_unref (buf=<optimized out>) at libavutil/buffer.c:139
#19 0x00007e27aa62bdbd in av_frame_unref (frame=0x7e2754451dc0) at libavutil/frame.c:618
#20 av_frame_unref (frame=0x7e2754451dc0) at libavutil/frame.c:610
#21 0x00007e27aa62bf1b in av_frame_free (frame=0x7e27830c8d50) at libavutil/frame.c:167
#22 0x00007e27aa63fbbe in vaapi_transfer_data_to (hwfc=<optimized out>, dst=0x7e275444c100, src=0x7e27544518c0) at libavutil/hwcontext_vaapi.c:1005
#23 0x00007e27aa63201a in av_hwframe_transfer_data (dst=0x7e275444c100, src=0x7e27544518c0, flags=flags@entry=0) at libavutil/hwcontext.c:481
#24 0x0000618063ee8db2 in mp_image_hw_upload (hw_img=0x7e2754450dd0, src=0x7e275476a000) at ../mpv/video/mp_image_pool.c:350
#25 mp_image_hw_upload (hw_img=0x7e2754450dd0, src=0x7e275476a000) at ../mpv/video/mp_image_pool.c:327
#26 mp_av_pool_image_hw_upload (hw_frames_ctx=<optimized out>, src=0x7e275476a000) at ../mpv/video/mp_image_pool.c:438
#27 0x0000618063f6a07f in try_format_upload (hw=hw@entry=0x7e27542c13f0, pixfmt=<optimized out>) at ../mpv/video/out/hwdec/hwdec_vaapi.c:436
#28 0x0000618063f6ab0c in try_format_config (hw=0x7e27542c13f0, hwconfig=0x7e27543424c0) at ../mpv/video/out/hwdec/hwdec_vaapi.c:504
#29 determine_working_formats (hw=0x7e27542c13f0) at ../mpv/video/out/hwdec/hwdec_vaapi.c:572
--Type <RET> for more, q to quit, c to continue without paging--c
#30 init (hw=0x7e27542c13f0) at ../mpv/video/out/hwdec/hwdec_vaapi.c:177
#31 0x0000618063ee2937 in ra_hwdec_load_driver
    (ra_ctx=<optimized out>, log=0x618080da77b0, global=<optimized out>, devs=0x7e27542c4a90, drv=0x618064071a80 <ra_hwdec_vaapi>, is_auto=false)
    at ../mpv/video/out/gpu/hwdec.c:104
#32 load_add_hwdec (ctx=0x7e27542bcad8, devs=0x7e27542c4a90, drv=0x618064071a80 <ra_hwdec_vaapi>, is_auto=false) at ../mpv/video/out/gpu/hwdec.c:236
#33 load_add_hwdec (ctx=0x7e27542bcad8, devs=0x7e27542c4a90, drv=0x618064071a80 <ra_hwdec_vaapi>, is_auto=<optimized out>) at ../mpv/video/out/gpu/hwdec.c:226
#34 0x0000618063eeaa62 in ra_hwdec_ctx_load_fmt (ctx=0x7e27542bcad8, devs=0x7e27542c4a90, params=0x7fff3baf5be0) at ../mpv/video/out/gpu/hwdec.c:332
#35 0x0000618063f05945 in gl_video_load_hwdecs_for_img_fmt (p=<optimized out>, devs=<optimized out>, params=0x7fff3baf5be0) at ../mpv/video/out/gpu/video.c:4369
#36 request_hwdec_api (vo=0x618080dc8960, data=0x7fff3baf5be0) at ../mpv/video/out/vo_gpu.c:134
#37 control (vo=0x618080dc8960, request=<optimized out>, data=0x7fff3baf5be0) at ../mpv/video/out/vo_gpu.c:203
#38 0x0000618063efadde in run_control (p=0x7fff3baf5b30) at ../mpv/video/out/vo.c:652
#39 0x0000618063e75cc2 in mp_dispatch_queue_process (queue=0x618080da30c0, timeout=<optimized out>) at ../mpv/misc/dispatch.c:300
#40 0x0000618063f0420e in vo_thread (ptr=<optimized out>) at ../mpv/video/out/vo.c:1100
#41 0x00007e27a990d39d in start_thread (arg=<optimized out>) at pthread_create.c:447
#42 0x00007e27a999249c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

### Do you want to contribute a patch to fix the issue?

None

database64128, 2024-12-04:
Same as https://github.com/intel/media-driver/issues/1879#issuecomment-2506602494.

q234rty, 2024-12-04:
libva_trace: 
[libva_trace.log.153023.thd-0x00008296.log](https://github.com/user-attachments/files/18010586/libva_trace.log.153023.thd-0x00008296.log)

intel-mediadev, 2024-12-05:
Auto Created VSMGWL-77367 for further analysis.

jiafengy1, 2024-12-05:
@q234rty   Could you share the complete cmd and video?

q234rty, 2024-12-05:
`mpv --no-config --hwdec=vaapi spbtv_sample_bipbop_av1_960x540_25fps.mp4` where the mp4 is the one from https://github.com/SPBTV/video_av1_samples/blob/master/spbtv_sample_bipbop_av1_960x540_25fps.mp4

jiafengy1, 2024-12-06:
Hi @q234rty  We can reproduce this issue on our side with the same driver version and gmmlib version. I have some questions:
1. What's the mpv version?
2. what's the libva version?
3. Is this issue always reproduced? or run to run reproduce?

q234rty, 2024-12-06:
mpv is on 0.39.0 and libva is on 2.22.0. For me, the issue is always reproducible.

jiafengy1, 2024-12-06:
> mpv is on 0.39.0 and libva is on 2.22.0. For me, the issue is always reproducible.

Could you share the repro environment?

q234rty, 2024-12-06:
> Could you share the repro environment?

Not sure what more can I share... The DE in use is KDE wayland if that's relevant. Have you tried reproducing using Arch Linux?

database64128, 2024-12-06:
> For me, the issue is always reproducible.

Same here. And I'm using GNOME wayland.

nyanmisaka, 2024-12-06:
One tip is that Arch linux has enabled **LTO by default** for the packages they build, which may not always be stable.

https://rfc.archlinux.page/0004-lto-by-default/

jiafengy1, 2024-12-06:
Hi, @q234rty  We have root caused this issue, and our fix patch is in the formal process.  @q234rty  @nyanmisaka  @database64128 Thanks for your support.

jiafengy1, 2024-12-13:
@q234rty  patch checked in, you can have a try.

nyanmisaka, 2024-12-13:
Can you also point out which commit contains the fix? This way users can patch it themselves before a new tag appears.

q234rty, 2024-12-14:
I confirm the current master works though I'm not sure which commit fixed this...

nyanmisaka, 2024-12-14:
Great, then at least 25.1.0 will be safe.

MicroYY, 2024-12-16:
https://github.com/intel/media-driver/commit/7faa3bfe1a4cb79180eae64ccb426af5102f11e3 is the fix commit.

jiafengy1, 2024-12-16:
@q234rty Now this issue has been resolved. Could you help close this issue?

q234rty, 2024-12-16:
I confirm patching 24.4.4 with just 7faa3bfe1a4cb79180eae64ccb426af5102f11e3 also fixes this issue.

closed 2024-12-16 by q234rty