intel/media-driver #1889: [Bug]: hevc_qsv: Error during QSV decoding.: GPU Hang (-21)

riaqn, 2024-12-07:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

```
ffmpeg -hwaccel qsv -c:v hevc_qsv -load_plugin hevc_hw -i /mnt/Downloads/movies/SomeMovie.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS/SomeMovie.BluRay.2160p.x
265.10bit.HDR.2Audio.mUHD-FRDS.mkv -vf hwdownload,format=p010 -pix_fmt p010le output.yuv -y -loglevel debug
```
prints:
```
ffmpeg version n7.1 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 14.2.1 (GCC) 20240910
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-frei0r --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libdvdnav --enable-libdvdread --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgsm --enable-libharfbuzz --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libplacebo --enable-libpulse --enable-librav1e --enable-librsvg --enable-librubberband --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-vapoursynth --enable-version3 --enable-vulkan
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.100 / 61. 19.100
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
Splitting the commandline.
Reading option '-hwaccel' ... matched as option 'hwaccel' (use HW accelerated decoding) with argument 'qsv'.
Reading option '-c:v' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'hevc_qsv'.
Reading option '-load_plugin' ... matched as AVOption 'load_plugin' with argument 'hevc_hw'.
Reading option '-i' ... matched as input url with argument '/mnt/Downloads/movies/007：大破天幕杀机.Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS/Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS.mkv'.
Reading option '-vf' ... matched as option 'vf' (alias for -filter:v (apply filters to video streams)) with argument 'hwdownload,format=p010'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'p010le'.
Reading option 'output.yuv' ... matched as output url.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option y (overwrite output files) with argument 1.
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /mnt/Downloads/movies/007：大破天幕杀机.Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS/Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS.mkv.
Applying option hwaccel (use HW accelerated decoding) with argument qsv.
Applying option c:v (select encoder/decoder ('copy' to copy stream without reencoding)) with argument hevc_qsv.
Successfully parsed a group of options.
Opening an input file: /mnt/Downloads/movies/007：大破天幕杀机.Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS/Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS.mkv.
[AVFormatContext @ 0x640afa738bc0] Opening '/mnt/Downloads/movies/007：大破天幕杀机.Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS/Skyfall.2012.BluRay.2160p.x265.10bit.HDR.2Audio.mUHD-FRDS.mkv' for reading
[file @ 0x640afa739280] Setting default whitelist 'file,crypto,data'
[matroska,webm @ 0x640afa738bc0] Format matroska,webm probed with size=2048 and score=100
st:0 removing common factor 1000000 from timebase
st:1 removing common factor 1000000 from timebase
st:2 removing common factor 1000000 from timebase
st:3 removing common factor 1000000 from timebase
st:4 removing common factor 1000000 from timebase
st:5 removing common factor 1000000 from timebase
[matroska,webm @ 0x640afa738bc0] Before avformat_find_stream_info() pos: 7504 bytes read:35090 seeks:2 nb_streams:6
[hevc @ 0x640afa74afc0] nal_unit_type: 32(VPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x640afa74afc0] Decoding VPS
[hevc @ 0x640afa74afc0] Main 10 profile bitstream
[hevc @ 0x640afa74afc0] nal_unit_type: 33(SPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x640afa74afc0] Decoding SPS
[hevc @ 0x640afa74afc0] Main 10 profile bitstream
[hevc @ 0x640afa74afc0] Decoding VUI
[hevc @ 0x640afa74afc0] nal_unit_type: 34(PPS), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x640afa74afc0] Decoding PPS
Transform tree:
    mdct_inv_float_avx2 - type: mdct_float, len: 32, factors[2]: [2, any], flags: [aligned, out_of_place, inv_only]
        fft16_asm_float_fma3 - type: fft_float, len: 16, factor: 2, flags: [aligned, inplace, out_of_place, preshuf, asm_call]
Transform tree:
[... 690 lines cut ...]
        RING_EXECLIST_SQ_CONTENTS: 0x0000000000000000
        RING_START: 0x0000000000000000
        RING_HEAD: 0x00000000
        RING_TAIL: 0x00000000
        RING_CTL: 0x00000000
        RING_MI_MODE: 0x00000200
        RING_MODE: 0x00000008
        RING_IMR: 0x00000000
        RING_ESR: 0x00000000
        RING_EMR: 0xffffffff
        RING_EIR: 0x00000000
        ACTHD: 0x0000000000000000
        BBADDR: 0x0000000000000000
        DMA_FADDR: 0x0000000000000000
        INDIRECT_RING_STATE: 0x00000000
        IPEHR: 0x00000000
        RING_INSTDONE: 0x7ffffffe

**** VM state ****
[0].error: -19
```

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

riaqn, 2024-12-07:
I want to note that:
- decoding h264 is fine
- encoding h264 is fine
- encoding hevc is also fine.

It's only decoding hevc that's problematic.

Jexu, 2024-12-16:
Looks that you are installing XE KMD on DG1 device, I think XE KMD is not our production support on dg1, pls try it again with I915 KMD.

riaqn, 2024-12-16:
@Jexu Actually I switched to XE because I915 is not working at all:
```
[vist#0:0/h264 @ 0x5c03d8c3a540] [dec:h264_qsv @ 0x5c03d8cba100] Error submitting packet to decoder: Input/output error
```
and I switched to XE as suggested by https://github.com/intel/media-driver/issues/1415#issuecomment-2225356256 . It was also suggested in that thread that DG1 is not supported by i915.

nyanmisaka, 2024-12-16:
A kernel driver regression about power gating caused this. Revert https://github.com/torvalds/linux/commit/38e8c4184ea0eebe478a9e6fc8d1062561223429

See also https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/3857

riaqn, 2024-12-16:
@nyanmisaka thank you for taking the efforts to pinpoint the issue! I couldn't have fixed it otherwise.

After downgrading linux kernel to 6.10.10, I confirm that with `xe` I can decode HEVC with QSV using ffmpeg.

Since this issue is unrelated to media-driver, I'm closing the issue.

closed 2024-12-16 by riaqn