intel/media-driver #1893: [Bug]: chromium crashes with intel-media-driver and Intel Arc B580

HeyItsBATMAN, 2024-12-17:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

None

### What happened?

When running Chromium, Chrome or anything using the CEF, the application will not start, always exiting with the following error:
`[5511:5511:1217/230941.026678:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230941.256342:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230942.024014:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230942.261787:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230943.009153:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230943.150395:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230943.915393:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230944.053984:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230944.792900:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230944.940533:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230945.751693:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230945.821455:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230946.200985:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230946.271470:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230946.660295:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[1217/230946.729769:ERROR:process_memory_linux.cc(50)] pread64: Input/output error (5)
[5511:5511:1217/230947.112485:ERROR:gpu_process_host.cc(982)] GPU process exited unexpectedly: exit_code=135
[5511:5511:1217/230947.112523:FATAL:gpu_data_manager_impl_private.cc(423)] GPU process isn't usable. Goodbye.`

### What's the usage scenario when you are seeing the problem?

Web browser

### Debug Information

I narrowed this down to either intel-media-driver or intel-gmmlib, so if anyone with more knowledge suspects this issue is due to gmmlib, I will post there instead.

I'm running Arch Linux, tested with Kernel Versions 6.12 and 6.13, Mesa 24.3 and Mesa 25.0.

Tested with package version: intel-media-driver 24.4.4.

GPU is the Intel Arc B580, with the 'xe' kernel driver module.

**lspci output**

```
$ lspci -k | grep -EA3 VGA
0a:00.0 VGA compatible controller: Intel Corporation Battlemage G21 [Intel Graphics]
	Subsystem: ASRock Incorporation Device 6021
	Kernel driver in use: xe
	Kernel modules: xe
```

**vainfo just crashes**

```
$ vainfo >vainfo.log 2>&1
fish: Job 1, 'vainfo >vainfo.log 2>&1' terminated by signal SIGBUS (Misaligned address error)
```

**package versions**

```
lib32-libva 2.22.0-1
libva 2.22.0-1.1
libva-intel-driver 2.4.1-3.1
libva-utils 2.22.0-1.1
intel-gmmlib 22.5.4-1.1
intel-media-driver 24.4.4-1.1
```

### Do you want to contribute a patch to fix the issue?

None

simonzzzzzzz, 2024-12-22:
I have exactly the same problem. I would provide further information as well if helpful. Thanks.

HeyItsBATMAN, 2024-12-22:
I'm unsure which stacktrace would be the most useful, but here is one captured using GDB when running vainfo.

**env**
```
LIBVA_DRIVERS_PATH=/usr/lib/dri/
LIBVA_DRIVER_NAME=iHD
```

**gdb backtrace vainfo**
```
$ gdb --batch --quiet --eval-command="run" --eval-command="backtrace" --eval-command="quit" vainfo
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Trying display: wayland

Program received signal SIGBUS, Bus error.
__memset_avx2_unaligned () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:150
150		VMOVU	%VMM(0), -VEC_SIZE(%rdi,%rdx)
#0  __memset_avx2_unaligned () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:150
#1  0x00007ffff5f4fc73 in XMHW_STATE_HEAP_INTERFACE::InitializeInterface (this=0x5555555f22b0, StateHeapSettings=...) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:1083
#2  Mhw_StateHeapInterface_Create (ppStateHeapInterface=<optimized out>, StateHeapSettings=...) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:680
#3  0x00007ffff5eff41f in mhw::render::Impl<mhw::render::xe2_hpg_next::Cmd>::AllocateHeaps (this=<optimized out>, stateHeapSettings=...) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/hw/mhw_render_impl.h:167
#4  0x00007ffff5e98c0e in XRenderHal_Platform_Interface_Next::AllocateHeaps (this=<optimized out>, pRenderHal=<optimized out>, MhwStateHeapSettings=...) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/renderhal/renderhal_platform_interface_next.cpp:807
#5  0x00007ffff5ea2bc7 in RenderHal_AllocateStateHeaps (pRenderHal=0x5555555ef870, pSettings=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/renderhal/renderhal.cpp:1499
#6  0x00007ffff5e9c45f in RenderHal_Initialize (pRenderHal=0x5555555ef870, pSettings=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/renderhal/renderhal.cpp:5352
#7  0x00007ffff5e866a7 in RenderCopyStateNext::Initialize (this=0x5555555ef310) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/shared/mediacopy/media_render_copy_next.cpp:119
#8  0x00007ffff5f36ccd in McpyDeviceXe2_Hpm::Initialize (this=0x5555555e3df0, osInterface=0x5555555caf30, mhwInterfaces=0x555555561de0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/media_interface/media_interfaces_bmg/media_interfaces_bmg.cpp:574
#9  0x00007ffff5ea5573 in McpyDeviceNext::CreateFactory (osDriverContext=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/media_interfaces/media_interfaces_next.cpp:425
#10 0x00007ffff569ab30 in MosMediaCopyBase::MosMediaCopyBase (this=0x5555555e6690, mosCtx=0x7fffffff6f70) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/os/mos_mediacopy_base.cpp:34
#11 MosMediaCopyBase::MosMediaCopyBase (this=0x5555555e6690, mosCtx=0x7fffffff6f70) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/os/mos_mediacopy_base.cpp:31
#12 MosMediaCopy::MosMediaCopy (this=0x5555555e6690, mosCtx=0x7fffffff6f70) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/os/mos_mediacopy.cpp:32
#13 MosUtilities::MosNewUtil<MosMediaCopy, _MOS_OS_CONTEXT*&> () at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/os/mos_utilities.h:3008
#14 OsContextSpecificNext::Init (this=0x5555555a7080, ddiDriverContext=0x7fffffff6f70) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/os/mos_context_specific_next.cpp:329
#15 0x00007ffff5f68034 in DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*, bool&) [clone .constprop.0] (ctx=<optimized out>, devicefd=<optimized out>, apoDdiEnabled=<optimized out>, minor_version=<optimized out>, major_version=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/ddi/media_libva.cpp:1801
#16 0x00007ffff582d582 in DdiMedia__Initialize (major_version=0x0, minor_version=0x0, ctx=0x555555562de0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/ddi/media_libva.cpp:1703
#17 __vaDriverInit (ctx=0x555555562de0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/ddi/media_libva.cpp:7528
#18 __vaDriverInit_1_22 (ctx=0x555555562de0) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/ddi/media_libva.cpp:7554
#19 0x00007ffff7f66064 in va_openDriver (dpy=0x555555562c70, driver_name=<optimized out>) at ../libva/va/va.c:478
#20 va_new_opendriver (dpy=0x555555562c70) at ../libva/va/va.c:714
#21 vaInitialize (dpy=0x555555562c70, major_version=0x7fffffffe2a4, minor_version=0x7fffffffe2a0) at ../libva/va/va.c:743
#22 vaInitialize (dpy=dpy@entry=0x555555562c70, major_version=major_version@entry=0x7fffffffe2a4, minor_version=minor_version@entry=0x7fffffffe2a0) at ../libva/va/va.c:727
#23 0x00005555555562b0 in main (argc=<optimized out>, argv=<optimized out>) at ../libva-utils-2.22.0/vainfo/vainfo.c:532
```

HeyItsBATMAN, 2024-12-23:
For anyone encountering the same issue: enable REBAR in your BIOS

closed 2024-12-23 by HeyItsBATMAN

DWSimmons, 2024-12-23:
> For anyone encountering the same issue: enable REBAR in your BIOS

Thank you for creating this issue/artifact.  Ubuntu 24.10 : kisak:mesa, intel B580 and couldn't get anywhere.  I checked bios and while rebar was set to Auto, my bios said to turn off Compatibility Security Module(CSM) to insure full rebar support.  I turned off CSM and my bios lost it's mind about Windows issues that I ignored due to Ubuntu.  Reboot and voila, all the hours of troubleshooting, driver swapping, command line, all good.  Appreciate it.

hadeedsarfaraz, 2025-11-22:
> what i should do if my bios doesn't support rebar??

I would take a look at https://github.com/xCuri0/ReBarUEFI