intel/media-driver #1896: [Bug]: Segmatation fault on B580 GPU when using vaapi with mpv

catcatmu, 2025-01-03:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

Decoding a video with mpv using  va-api on a B580 GPU caused a segmentation fault:
Command used: `mpv --hwdec=vaapi` or  `mpv --hwdec=auto` 

```
vp::VpAllocator::DestroyVpSurface (this=0x7fffb8467db0, surface=@0x7fffd8fef178: 0x7fffb841cb90, deferredDestroyed=false, flags=...) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508                                    
508             int64_t currentSize = static_cast<int64_t>(surface->osSurface->OsResource.pGmmResInfo ? surface->osSurface->OsResource.pGmmResInfo->GetSizeAllocation() : 0);
```

Video codec (reported by vlc was) `Codec: MPEG-H Part2/HEVC (H.265) (hevc)`  and `H264 - MPEG-4 AVC (part 10) (avc1)`

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

Video playback with hardware acceleration 

### Debug Information

#### Versions
|Component|Version|
|--|--|
|libva|2.22.0-1|
|libva-utils|2.22.0-1|
|intel-gmmlib|22.5.4-1|
|intel-media-driver|24.4.4-1|
|kernel: 6.12.7|arch1-1|

#### Hardware exists
Yes: `card0  renderD128`

#### Hardware Info
0b:00.0 VGA compatible controller [0300]: Intel Corporation Battlemage G21 [Intel Graphics] [8086:e20b]

#### VaInfo
[vainfo.log](https://github.com/user-attachments/files/18304266/vainfo.log)

### LibVa Trace
[va-trace.log](https://github.com/user-attachments/files/18304273/va-trace.log)

### Application Backtrace
[backtrace.txt](https://github.com/user-attachments/files/18304344/backtrace.txt)

### Dmesg Info
No Gpu messages
`segfault at 7adf7c433 ip 00007adf61f85698 sp 00007adf87ff50f0 error 4 in iHD_drv_video.so[585698,7adf61a1e000+986000] likely on CPU 22 (core 6, socket 0)`

### Do you want to contribute a patch to fix the issue?

None

database64128, 2025-01-06:
Probably same as #1887? Build from master and see if it's fixed for you.

catcatmu, 2025-01-06:
#1887  is the same issue. A build from master fixes the issue.

closed 2025-01-06 by catcatmu