intel/media-driver #1903: [Feature]: Add ARM64 platfrom support for DG2

matalama80td3l, 2025-01-24:
### What Feature?

Starting from Linux 6.8, the DG2 graphics card can work on the arm64 platform using the new xe kernel driver. However, the media-driver driver relies on the x86 instruction set, which prevents the DG2 graphics card from performing video codec on the arm64 platform.

As a C/C++ noob, I tried to add [sse2neon](https://github.com/DLTcollab/sse2neon) header to the project and remove some of the x86 restrictions but I receive a link error in the final stage of compilation. 

### What's the usage scenario would be benifited?

Transcode for media delivery

### Do you want to contribute a patch to develop this feature?

None

matalama80td3l, 2025-03-18:
I have finished it. I will upload a fork repo later. But I won't commit it to master branch because the code is a bit dirty and not working with DG2(only working with DG1).

nyanmisaka, 2025-03-18:
Xe KMD does not implement HuC firmware support on DG2. This means that media functionality is broken.

https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/234

matalama80td3l, 2025-03-19:
> Xe KMD does not implement HuC firmware support on DG2. This means that media functionality is broken.
> 
> https://gitlab.freedesktop.org/drm/xe/kernel/-/issues/234

我知道这个，但是感觉问题可能没到那个层次，vainfo直接就报bus error了，没有在x86平台上用xe kmd试过。

nyanmisaka, 2025-03-19:
Until DG2 gets HuC firmware support from Xe KMD, any user-side efforts will lead to dead end.