intel/media-driver #1905: [Bug]: vaapi --hwdec=vaapi

codecnotsupported, 2025-01-28:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

Might be caused by mpv.

Using:
`mpv --hwdec=vaapi ` with Battlemage

Logs:

[libva_trace.log](https://github.com/user-attachments/files/18567334/libva_trace.log)

BackTrace:
```
thread 12 "vo" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd6dfd6c0 (LWP 200107)]
vp::VpAllocator::DestroyVpSurface (this=0x7fffb0794310, surface=@0x7fffd6df30f8: 0x7fffb07bac50, deferredDestroyed=false, flags=...)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
508             int64_t currentSize = static_cast<int64_t>(surface->osSurface->OsResource.pGmmResInfo ? surface->osSurface->OsResource.pGmmResInfo->GetSizeAllocation() : 0);
(gdb) bt
#0  vp::VpAllocator::DestroyVpSurface (this=0x7fffb0794310, surface=@0x7fffd6df30f8: 0x7fffb07bac50, deferredDestroyed=false, flags=...)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/bufferMgr/vp_allocator.cpp:508
#1  0x00007fff95793793 in vp::SwFilterPipe::Clean (this=<optimized out>) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/feature_manager/sw_filter_pipe.cpp:409
#2  0x00007fff95b51903 in vp::VpObjAllocator<vp::SwFilterPipe>::Destory (this=0x7fffb017b670, obj=@0x7fffb07f3bd0: 0x7fffb07ec180)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.h:84
#3  vp::SwFilterPipeFactory::Destory (this=0x7fffb017b668, swFilterPipe=@0x7fffb07f3bd0: 0x7fffb07ec180)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/feature_manager/vp_obj_factories.cpp:298
#4  vp::VpPipeline::ExecuteSingleswFilterPipe(vp::VpSinglePipeContext*, vp::SwFilterPipe*&, vp::PacketPipe*, vp::VpFeatureManagerNext*) [clone .constprop.0] (this=this@entry=0x7fffb056e910, singlePipeCtx=<optimized out>, 
    pipe=@0x7fffb07f3bd0: 0x7fffb07ec180, featureManagerNext=featureManagerNext@entry=0x7fffb07acc70, pPacketPipe=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:713
#5  0x00007fff957e8381 in vp::VpPipeline::ExecuteVpPipeline (this=0x7fffb056e910) at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline.cpp:610
#6  0x00007fff9580f5a5 in virtual thunk to VpPipelineAdapterXe2_Hpm::Execute(VPHAL_RENDER_PARAMS*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/Xe_M_plus/Xe2_HPM/vp/hal/pipeline/vp_pipeline_adapter_xe2_hpm.h:73
#7  0x00007fff957e8a98 in VpPipelineAdapter::Render (this=0x7fffb07ee448, pcRenderParams=0x7fffb07bcaa0)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/common/vp/hal/pipeline/vp_pipeline_adapter.cpp:177
#8  0x00007fff9581540b in virtual thunk to VpPipelineAdapterXe2_Hpm::Render(VPHAL_RENDER_PARAMS const*) ()
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/agnostic/Xe_M_plus/Xe2_HPM/vp/hal/pipeline/vp_pipeline_adapter_xe2_hpm.h:53
#9  0x00007fff9548bf85 in DdiVpFunctions::EndPicture (this=<optimized out>, ctx=0x7fffb047a3f0, context=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:534
#10 0x00007fff95490541 in DdiVpFunctions::ProcessPipeline (this=0x7fffb05250b0, vaDrvCtx=0x7fffb047a3f0, ctxID=3489660928, srcSurface=2, srcRect=0x7fffd6df3b68, dstSurface=1, dstRect=0x7fffd6df3b70)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/vp/ddi/ddi_vp_functions.cpp:4472
#11 0x00007fff95449f0a in MediaLibvaInterfaceNext::PutImage (ctx=0x7fffb047a3f0, surface=1, image=<optimized out>, srcX=0, srcY=0, srcWidth=2, srcHeight=2, destX=0, destY=0, destWidth=2, destHeight=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_softlet/linux/common/ddi/media_libva_interface_next.cpp:1867
#12 MediaLibvaInterface::PutImage (ctx=0x7fffb047a3f0, surface=1, image=<optimized out>, src_x=0, src_y=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-24.4.4/media_driver/linux/common/ddi/media_libva_interface.cpp:384
#13 0x00007ffff41b583f in vaPutImage (dpy=0x7fffb047a280, surface=surface@entry=1, image=0, src_x=src_x@entry=0, src_y=src_y@entry=0, src_width=2, src_height=2, dest_x=0, dest_y=0, dest_width=2, dest_height=2)
    at ../libva/va/va.c:1893
#14 0x00007ffff4e3c37d in vaapi_unmap_frame (hwfc=0x7fffb07f8900, hwmap=<optimized out>) at libavutil/hwcontext_vaapi.c:782
#15 0x00007ffff4e31dc3 in ff_hwframe_unmap (opaque=<optimized out>, data=0x7fffb0530200 "\300\002S\260\377\177") at libavutil/hwcontext.c:717
#16 0x00007ffff4e1c9b7 in buffer_replace (dst=<optimized out>, src=0x0) at libavutil/buffer.c:133
#17 av_buffer_unref (buf=<optimized out>) at libavutil/buffer.c:144
#18 av_buffer_unref (buf=<optimized out>) at libavutil/buffer.c:139
#19 0x00007ffff4e2bdbd in av_frame_unref (frame=0x7fffb052ae40) at libavutil/frame.c:618
#20 av_frame_unref (frame=0x7fffb052ae40) at libavutil/frame.c:610
#21 0x00007ffff4e2bf1b in av_frame_free (frame=0x7fffd6df3d10) at libavutil/frame.c:167
#22 0x00007ffff4e3fbbe in vaapi_transfer_data_to (hwfc=<optimized out>, dst=0x7fffb0526280, src=0x7fffb052aac0) at libavutil/hwcontext_vaapi.c:1005
#23 0x00007ffff4e3201a in av_hwframe_transfer_data (dst=0x7fffb0526280, src=0x7fffb052aac0, flags=flags@entry=0) at libavutil/hwcontext.c:481
#24 0x0000555555637db2 in mp_image_hw_upload (hw_img=0x7fffb07f9890, src=0x7fffb052ffc0) at ../mpv/video/mp_image_pool.c:350
#25 mp_image_hw_upload (hw_img=0x7fffb07f9890, src=0x7fffb052ffc0) at ../mpv/video/mp_image_pool.c:327
#26 mp_av_pool_image_hw_upload (hw_frames_ctx=<optimized out>, src=0x7fffb052ffc0) at ../mpv/video/mp_image_pool.c:438
--Type <RET> for more, q to quit, c to continue without paging--c
#27 0x00005555556b907f in try_format_upload (hw=hw@entry=0x7fffb0413120, pixfmt=<optimized out>) at ../mpv/video/out/hwdec/hwdec_vaapi.c:436
#28 0x00005555556b9b0c in try_format_config (hw=0x7fffb0413120, hwconfig=0x7fffb0441f40) at ../mpv/video/out/hwdec/hwdec_vaapi.c:504
#29 determine_working_formats (hw=0x7fffb0413120) at ../mpv/video/out/hwdec/hwdec_vaapi.c:572
#30 init (hw=0x7fffb0413120) at ../mpv/video/out/hwdec/hwdec_vaapi.c:177
#31 0x0000555555631937 in ra_hwdec_load_driver (ra_ctx=<optimized out>, log=0x55555589d280, global=<optimized out>, devs=0x7fffb0413600, drv=0x5555557c0a80 <ra_hwdec_vaapi>, is_auto=false) at ../mpv/video/out/gpu/hwdec.c:104
#32 load_add_hwdec (ctx=0x7fffb040af68, devs=0x7fffb0413600, drv=0x5555557c0a80 <ra_hwdec_vaapi>, is_auto=false) at ../mpv/video/out/gpu/hwdec.c:236
#33 load_add_hwdec (ctx=0x7fffb040af68, devs=0x7fffb0413600, drv=0x5555557c0a80 <ra_hwdec_vaapi>, is_auto=<optimized out>) at ../mpv/video/out/gpu/hwdec.c:226
#34 0x0000555555639a62 in ra_hwdec_ctx_load_fmt (ctx=0x7fffb040af68, devs=0x7fffb0413600, params=0x7fffffffd970) at ../mpv/video/out/gpu/hwdec.c:332
#35 0x0000555555654945 in gl_video_load_hwdecs_for_img_fmt (p=<optimized out>, devs=<optimized out>, params=0x7fffffffd970) at ../mpv/video/out/gpu/video.c:4369
#36 request_hwdec_api (vo=0x55555597a320, data=0x7fffffffd970) at ../mpv/video/out/vo_gpu.c:134
#37 control (vo=0x55555597a320, request=<optimized out>, data=0x7fffffffd970) at ../mpv/video/out/vo_gpu.c:203
#38 0x0000555555649dde in run_control (p=0x7fffffffd8c0) at ../mpv/video/out/vo.c:652
#39 0x00005555555c4cc2 in mp_dispatch_queue_process (queue=0x5555558c9980, timeout=<optimized out>) at ../mpv/misc/dispatch.c:300
#40 0x000055555565320e in vo_thread (ptr=<optimized out>) at ../mpv/video/out/vo.c:1100
#41 0x00007ffff40052ce in start_thread (arg=<optimized out>) at pthread_create.c:447
#42 0x00007ffff408a29c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

### What's the usage scenario when you are seeing the problem?

Playback

### Do you want to contribute a patch to fix the issue?

None

database64128, 2025-02-01:
I think it's the same as #1896.

codecnotsupported, 2025-02-01:
> I think it's the same as [#1896](https://github.com/intel/media-driver/issues/1896).

Yes, it indeed works fine with the git branch.
For arch users, simply run. And let it compile.
`yay -S intel-media-driver-git`

closed 2025-02-01 by codecnotsupported