intel/media-driver #1909: [Bug]: MFX_ERR_GPU_HANG  when encode  1*UHD 444 10bit HEVC + 3 * FHD 420 8bit H264

okfujian, 2025-02-20:
### Is it regression? Good in old configuration?

None

### What happened?

when we use 11th Gen Intel(R) Core(TM) i7-1165G7,  2024Q4 ONEVPL release (https://github.com/intel/vpl-gpu-rt/releases/tag/intel-onevpl-24.4.4).      I have commit this issue to github.    
    I met a issue:     
when encode  1*UHD 444 10bit HEVC + 3 * FHD 420 8bit H264   sample_encode return MFX_ERR_GPU_HANG  .  

  /opt/intel/media/share/vpl/samples/_bin/sample_encode h265 -gpucopy::on -y410 -ec::y410 -i /tmp/yuv/test.yuv -o 444.h265 -w 3840 -h 2160 -dstw 3840 -dsth 2160 -b 10000 -hw  -async 2 -lowpower:on -timeout 600 -u speed -fps 60 -f 60  
/opt/intel/media/share/vpl/samples/_bin/sample_encode h264 -gpucopy::on  -i /tmp/yuv/nv12.yuv -o test1.h265 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -hw  -async 2 -lowpower:on -timeout 60 -u speed -fps 60 -f 60  &
/opt/intel/media/share/vpl/samples/_bin/sample_encode h264 -gpucopy::on  -i /tmp/yuv/nv12.yuv -o test2.h265 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -hw  -async 2 -lowpower:on -timeout 60 -u speed -fps 60 -f 60  &
/opt/intel/media/share/vpl/samples/_bin/sample_encode h264 -gpucopy::on  -i /tmp/yuv/nv12.yuv -o test3.h265 -w 1920 -h 1080 -dstw 1920 -dsth 1080 -b 10000 -hw  -async 2 -lowpower:on -timeout 60 -u speed -fps 60 -f 60  &

os version:ubuntu 20.04   

5.15.0-56-generic

attachment is /sys/kernel/debug/dri/0/i915_error_state

this is our igfx_user_feature:
[image]

[image]

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

None

Sherry-Lin, 2025-02-20:
Thanks for reporting this issue. That would be great if you could provide more information like
1. Is it related to 1+3 multi-encode specifically? Any issues found if only 1 or only 3 encode?
2. Is it related to 444 or big resolution or not? Any issues found if encode 420 or lower resolution?

okfujian, 2025-02-24:
"Is it related to 1+3 multi-encode specifically? Any issues found if only 1 or only 3 encode?  "            only 1 encode or only 3 encode don't have issue.

"Is it related to 444 or big resolution or not? Any issues found if encode 420 or lower resolution? “    I think related to 444 10bit,  when 4K 420 10bit, don't have issue.

okfujian, 2025-09-18:
@Jexu  @XinfengZhang   can you help check this issue, this is very important for our company.