intel/media-driver #1916: [Bug]: Non-X86 build Error

wx5391805, 2025-03-15:
### Is it regression? Good in old configuration?

None

### What happened?

I'm building a driver on Android aarch64 platform. Why it requires SSE2/SSE4 now? Can this library run on non-x86 machines?

We can see it in `cm_mem_sse2_impl.cpp`:
```
#if defined(__SSE2__) || !(defined(LINUX) || defined(ANDROID))
```
Why Android requires it?

My Errors:
```
In file included from ../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:27:
In file included from ../media_driver/agnostic/common/cm/cm_mem.h:28:
/home/user/android/android-ndk-r26d/toolchains/llvm/prebuilt/linux-x86_64/lib/clang/17/include/mmintrin.h:14:2: error: "This header is only meant to be used on x86 and x64 architecture"
#error "This header is only meant to be used on x86 and x64 architecture"

```

### What's the usage scenario when you are seeing the problem?

Cloud Gaming

### Do you want to contribute a patch to fix the issue?

None

matalama80td3l, 2025-03-18:
well, you have to use something like sse2neon to replace those x86 instruction to aarch64 forms. I have successfully done this a month ago, but it not work with dg2, only work with DG1.

wx5391805, 2025-03-18:
> well, you have to use something like sse2neon to replace those x86 instruction to aarch64 forms. I have successfully done this a month ago, but it not work with dg2, only work with DG1.

Thank you for your information, I'll try it but why is DG2 not working?  I need to make DG2 work.

matalama80td3l, 2025-03-19:
I don't know. Maybe because of this:https://github.com/intel/media-driver/issues/1903#issuecomment-2733596237

wx5391805, 2025-03-19:
That makes the problem seem obvious—after a year, DG2 still doesn't have HuC support, while DG1 does. That's a bit unsettling.