intel/media-driver #1918: [Bug]: Build with cmake-4.0.0 fails

heitbaum, 2025-03-28:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

None

### What happened?

Build with cmake-4.0.0
```
CMake Error at os_release_info.cmake:32 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Build of media-driver

### Do you want to contribute a patch to fix the issue?

Yes, I'm glad to submit a patch to fix it

heitbaum, 2025-03-28:
The CMakeLists.txt fiules in the repository have a varying version requirements. But a minimum of 3.5 is in many of them. Fix will be to update all the CMakeLists.txt to a minimum of 3.5 (or ideally 3.5...3.10 to address the future deprecation of CMake 3.5

https://github.com/search?q=repo%3Aintel%2Fmedia-driver%20cmake_minimum_required&type=code

fkobi, 2026-10-14:
https://github.com/intel/media-driver/commit/04584f8839ff773f18e4b1542618a00b6fc4de1a and https://github.com/intel/media-driver/blob/11ecb9ea0add68b9a1439a31c28af8e97ab65a2c/CMakeLists.txt#L21 make me believe that this is coming soon

tom--pollard, 2026-10-20:
Yep looks like this can be closed now as of https://github.com/intel/media-driver/releases/tag/intel-media-26.3.5

heitbaum, 2026-10-20:
> 10\. Yep looks like this can be closed now as of https://github.com/intel/media-driver/releases/tag/intel-media-26.3.5

There as still 9 references to previous cmake versions. #1919

tom--pollard, 2026-10-21:
> 11\. > 10. Yep looks like this can be closed now as of https://github.com/intel/media-driver/releases/tag/intel-media-26.3.5
>      
>      
>      There as still 9 references to previous cmake versions. [#1919](https://github.com/intel/media-driver/pull/1919)

Fair yeh, not universally across the codebase there's still some tools not converted