intel/media-driver #1926: [Bug]: MPV crashed on LNL

crab2313, 2025-05-04:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

None

### What happened?

1. open a video with MPV
2. seek forward with the right arrow key
3. the MPV will crash randomly

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

This is backtrace on latest archlinux:

```
(gdb) bt
#0  MediaStatusReport::GetCompletedCount (this=0x0, this=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-25.2.1/media_softlet/agnostic/common/shared/statusreport/media_status_report.h:174
#1  decode::AvcPipelineXe2_Lpm_Base::GetCompletedReport (this=0x5ba17fbc1a60)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-25.2.1/media_softlet/agnostic/Xe2_M_plus/Xe2_LPM_base/codec/hal/dec/avc/pipeline/decode_avc_pipeline_xe2_lpm_base.cpp:77
#2  DecodeAvcPipelineAdapterXe2_Lpm_Base::GetCompletedReport (this=0x5ba17fd8d280)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-25.2.1/media_softlet/agnostic/Xe2_M_plus/Xe2_LPM_base/codec/hal/dec/avc/pipeline/decode_avc_pipeline_adapter_xe2_lpm_base.cpp:124
#3  0x00007b9bf94a308f in DdiDecodeFunctions::StatusReport
    (this=this@entry=0x7b9bf417c230, mediaCtx=mediaCtx@entry=0x7b9bf401e9b0, decoder=0x5ba17fd8d280, surface=surface@entry=0x5ba17fc8f2b0)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-25.2.1/media_softlet/linux/common/codec/ddi/dec/ddi_decode_functions.cpp:1049
#4  0x00007b9bf94a3244 in DdiDecodeFunctions::StatusCheck
    (this=0x7b9bf417c230, mediaCtx=0x7b9bf401e9b0, surface=0x5ba17fc8f2b0, surfaceId=<optimized out>)
    at /usr/src/debug/intel-media-driver/media-driver-intel-media-25.2.1/media_softlet/linux/common/codec/ddi/dec/ddi_decode_functions.cpp:1031
#5  0x00007b9c4027cced in vaSyncSurface (dpy=0x7b9bf4002620, render_target=<optimized out>) at ../libva/va/va.c:1668
#6  0x00005ba15fc4f929 in ??? ()
#7  0x00005ba15fbc8d9a in ??? ()
#8  0x00005ba15fbee26f in ??? ()
#9  0x00007b9c43ff17e0 in acquire_frame (pass=0x7b9c22bf0de0, frame=0x7b9c22bf0f48, acquired=0x7b9c22bf1af2) at src/renderer.c:2881
#10 pass_init (pass=pass@entry=0x7b9c22bf0de0, acquire_image=acquire_image@entry=true) at src/renderer.c:3014
#11 0x00007b9c43ff9111 in pl_render_image
    (rr=rr@entry=0x7b9bf4043710, pimage=pimage@entry=0x7b9bf4189570, ptarget=ptarget@entry=0x7b9c22bf4f90, params=params@entry=0x7b9c22bf4b90)
    at src/renderer.c:3138
#12 0x00007b9c43ffa870 in pl_render_image_mix (rr=<optimized out>, images=<optimized out>, ptarget=0x7b9c22bf4f90, params=0x7b9c22bf4b90)
    at src/renderer.c:3688
#13 0x00005ba15fbf0f11 in ??? ()
#14 0x00005ba15fbe7d1d in ??? ()
#15 0x00007b9c400d27eb in start_thread (arg=<optimized out>) at pthread_create.c:448
#16 0x00007b9c4015618c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

### Do you want to contribute a patch to fix the issue?

No.

database64128, 2025-05-27:
I think it's the same issue as #1879, and it affects all platforms starting from Meteor Lake. It's a shame that our reports are just ignored by Intel.

ye-discernis, 2026-09-24:
This looks like the same bug as #1879 (same `vaSyncSurface → StatusReport → GetCompletedReport → GetCompletedCount(this=0x0)` path). I reproduced it on Arrow Lake-HX with AV1, and my core dump shows the likely cause: at the moment of the crash, mpv's main thread is inside `vaDestroyContext` for the same decoder (`avcodec_free_context` → libavcodec vaapi uninit). So the pipeline's `m_statusReport` is torn down while the VO thread is still syncing one of its surfaces. That fits the "keep seeking" repro here, because each seek can free the decoder while a frame is still in flight.

Both thread stacks and details: https://github.com/intel/media-driver/issues/1879#issuecomment-5823263516

Workaround until it's fixed: `hwdec=vaapi-copy`. Decoding stays on the GPU, but the VO thread never calls `vaSyncSurface`, so the race can't happen.