intel/media-driver #1927: SIGBUS crash in RenderHal_Initialize using VA-API with Arc B580 (Battlemage G21)

butchermustdie, 2025-05-04:
**Describe the bug**

Running `vainfo` on Intel Arc B580 (Battlemage G21) with the `iHD` driver results in a **SIGBUS (Bus error)** crash during heap initialization in `RenderHal_Initialize`.

The issue appears in `memset()` during state heap interface setup.

---

**To Reproduce**

Steps to reproduce the behavior:

1. Use Ubuntu 25.04 with latest kernel and Mesa stack.
2. Install `intel-media-driver` with `iHD` support.
3. Ensure `LIBVA_DRIVER_NAME=iHD` and run:

   ```bash
   vainfo
   ```
4. Observe a SIGBUS crash.

---

**Expected behavior**

`vainfo` should detect and list supported VA-API profiles using the iHD driver.

---

**Stack trace (from GDB)**

```
Program received signal SIGBUS, Bus error.
__memset_avx2_unaligned_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:210

#0  __memset_avx2_unaligned_erms ()
#1  0x00007ffff5e45973 in XMHW_STATE_HEAP_INTERFACE::InitializeInterface (
    this=0x555555604230, StateHeapSettings=...)
    at /media-driver/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:1083
#2  0x00007ffff5e45d27 in Mhw_StateHeapInterface_Create (
    ppStateHeapInterface=<optimised out>, StateHeapSettings=...)
    at /media-driver/media_softlet/agnostic/common/hw/mhw_state_heap.cpp:680
#3  0x00007ffff5d8676b in mhw::render::Impl<mhw::render::xe2_hpg_next::Cmd>::AllocateHeaps (this=<optimised out>, stateHeapSettings=...)
    at /media-driver/media_softlet/agnostic/common/hw/mhw_render_impl.h:167
#4  0x00007ffff5d39dcf in XRenderHal_Platform_Interface_Next::AllocateHeaps (
    this=<optimised out>, pRenderHal=<optimised out>, MhwStateHeapSettings=...)
    at /media-driver/media_softlet/agnostic/common/renderhal/renderhal_platform_interface_next.cpp:807
#5  0x00007ffff5d36b04 in RenderHal_AllocateStateHeaps (
    pRenderHal=0x5555556017b0, pSettings=0x555555601858)
    at /media-driver/media_softlet/agnostic/common/renderhal/renderhal.cpp:1499
#6  0x00007ffff5d35005 in RenderHal_Initialize (pRenderHal=0x5555556017b0, ...
```

---

**Environment**

* **GPU**: Intel Arc B580 (Battlemage G21)
* **Kernel**: 6.15.0-rc4+
* **OS**: Ubuntu 25.04 (plucky)
* **Media driver version**: latest build from source (May 4, 2025)
* **libva version**: 2.22.0
* **vainfo**: crashes with `SIGBUS`
* **Mesa version**: 25.0.3

---

**Additional context**

* The crash happens regardless of Wayland/X11.
* Mesa VA driver (`mesa_drv_video.so`) does not support this hardware.
* User is in `render` group, permissions are correct.

---

**Suggested label**: `bug`, `Battlemage`, `Arc`, `SIGBUS`, `heap allocation`, `VAAPI`

---

nyanmisaka, 2026-01-14:
Hi @XinfengZhang, could you please clarify whether both Battlemage and future dGPUs using Xe KMD will no longer be compatible with motherboards that do not support **resizable BAR**?

We have encountered many similar cases on Arc B580/B570 and Pro B50, and if this is indeed the case, I would advise them to replace their motherboard or find an alternative. It would be even better if this limitation were mentioned in the README.

butchermustdie, 2026-01-14:
Hi!
The issue was resolved only by enabling ReBAR. To do this, I had to reflash the motherboard BIOS.

https://github.com/Koshak1013/HuananzhiX99_BIOS_mods?tab=readme-ov-file#%D0%B4%D0%BE%D0%B1%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5-%D0%BF%D0%BE%D0%B4%D0%B4%D0%B5%D1%80%D0%B6%D0%BA%D0%B8-resizable-bar-%D0%B2-%D0%B1%D0%B8%D0%BE%D1%81
https://forums.overclockers.ru/viewtopic.php?f=1&t=602922&start=42740